AOSP-JF-MM/issue_tracking_aosp #5: No audio recorded in most apps

asfaltboy, 2015-12-04:
It seems that all applications except the phone app capture no audio. Some applications (skype, hangouts) request the permission to record audio, but even after allowing those nothing is recorded.

Not sure which logs to provide here, but this feels quite general.

B--B, 2015-12-05:
Fixed with this commit: https://github.com/AOSP-JF-MM/platform_frameworks_av/commit/91531d12a1f6b4546abdbf96a2fc81d6583e3eb7

closed 2015-12-05 by B--B

asfaltboy, 2015-12-06:
Thanks @arco. I wonder, do you guys have a automated build flow that produces a ROM image for every commit merged to master? IOW is there an easy way to get a version to test resolution?

arco, 2015-12-06:
Hey,

Not that I know of, but I think we can initiate a build manually. I build and test locally though. Normally a full build isn't needed, but just the component you want to test, by pushing it to the device and rebooting.

asfaltboy, 2015-12-06:
Thanks! Is there a guide you could point me to, in order to understand the build steps of specific AOSP-JF-MM component (I assume such as [platform_frameworks_av](https://github.com/AOSP-JF-MM/platform_frameworks_av))?

javelinanddart, 2015-12-06:
You'd first have to understand building a ROM fully from source

fafualex, 2015-12-08:
Having the same problem: "It seems that all applications except the phone app capture no audio."
I saw that @arco fixed the problem 3 days ago in his commit.. I'm not able to build myself the rom for testing it any tips and tricks for noobs? ;) thanks in advance. 

Phone model: GT-I9505
Baseband version: I9505XXUDMGG
ROM version: 13.0-20151108-UNOFFICIAL-jflte
GApps Package: open_gapps-arm-6.0-full-20151113
Did you do a clean install for the currently installed build? yes
Did you reboot after having the issue? yes
Did you do any system modification? no
Did you flash another kernel? no
Did you install Xposed framework? yes

javelinanddart, 2015-12-08:
1. Don't used Xposed
2. If you look on XDA there are plenty of guides on how to build a ROM

fafualex, 2015-12-08:
I've installed xposed after having flashed the rom for the second time and saw if the microphone problem was still existing. So xposed doesn't change nothing for this problem..

asfaltboy, 2015-12-08:
I can confirm that latest From (6th Dec) solves the issue

Best regards,
Pavel Savchenko

fafualex, 2015-12-08:
Ok I saw now that they published the latest build on 6th December! Thanks a lot! ;)

asfaltboy, 2015-12-08:
I can confirm that the latest ROM ([from 6th Dec](http://forum.xda-developers.com/showpost.php?p=64178560&postcount=407)) resolves the issue.

fafualex, 2015-12-08:
The microphone fix it's working for me too!! =D really great job! very smooth animation and other cool stuff! 
I just can't understand one thing, I can't find the theme chooser app but instead I've found Layers Manager app that it's a little bit laggy and I can't remove the theme I've applied. 

I've installed "open_gapps-arm-6.0-full-20151208" and I see some other apps like message, file explorer and browser (material design I've found on the first cm13) are disappeared.

smeroni68, 2015-12-08:
On AOSP there is no Theme Engine (CM), but Layer Manager with RRO layers.
Layer Manager let you use also different combinations of layers from different packages. Until now, the removing of layers is not working, probably for a permission on folders. If you want remove them or some of them, you can go with a root explorer on root folder on vendor/overlay, mount ar R/W and delete manually the layer files (just remember to soft reboot after deleting.
If you update the rom, overlays are automatically deleted and you must re-apply.

fafualex, 2015-12-09:
Thanks you a lot man!!! 😉 you've saved me.. It worked fine and after reboot everything was on stock theme! 👍