AOSP-JF-MM/issue_tracking_aosp #6: No touch and SIM detected when device is starting connected to charger

asfaltboy, 2015-12-06:
When my device starts up and the USB cable is connected to a charger the device boots, however the touch is unresponsive (and mechanical buttons *are responsive*), and the status bar shows "no sim" message.

Phone model: GT-I9505
Baseband version: I9505XXUEML1
ROM version: 20151108
GApps Package: gapps marshmallow mini (from the instructions)

Did you do a clean install for the currently installed build? yes
Did you reboot after having the issue? yes
Did you do any system modification? no
Did you flash another kernel? no
Did you install Xposed framework? no

javelinanddart, 2015-12-06:
Don't leave charger plugged in when booting up, it's been known to cause issues on Samsung Devices

asfaltboy, 2015-12-06:
I seem to recall reading it somewhere, but did no if it had any basis. In any case, this happens to me every time power runs out in my battery and I have no spare battery on hand. Ever since I moved to AOSP 6 there seems to have been a constant power drain, so this happens to me more often.

closed 2015-12-31 by B--B