AOSP-JF-MM/issue_tracking_aosp #9: No Caller ID on VZW

FernBch, 2015-12-07:
Caller ID on Verizon is not working, either.

Logs:

[url]https://www.dropbox.com/s/311buo7oevpt9ms/2015-12-06_19.10.zip?dl=0[/url]

[IMG]http://i1075.photobucket.com/albums/w434/FernBchJeep/Mobile%20Uploads/Screenshot_20151206-190941_zps9c6biqau.png[/IMG]

javelinanddart, 2015-12-07:
@invisiblek Fixed it a while back

FernBch, 2015-12-07:
Could I be missing something? It hasn't worked for me since leaving. AOSP 5.1.

FernBch, 2015-12-07:
Edit:

Could it possibly be my variant? I know in the past there were other things that were somwtimes "screwy" on the I545. The compass issues from CM11 were a good example.

javelinanddart, 2015-12-07:
Dan has I545, Maybe the commit hasn't been picked
https://github.com/CyanogenMod/android_device_samsung_jf-common/commit/c349c7333c08560ad2576cd45a5392f2d9840089

javelinanddart, 2015-12-07:
Or RIL could be screwy on AOSP

FernBch, 2015-12-07:
Back in the days of CM11 someone posted in the thread he no longer uses it. Is there anything else I can provide? There are others on xda saying the same thing.

invisiblek, 2015-12-07:
Caller ID is/was working fine last I tried on CM. Are we using the same blobs?

javelinanddart, 2015-12-07:
That could also be it, We might not be on the same blobs.
It doesn't look like we are.

invisiblek, 2015-12-07:
Yea, that's almost definitely the issue then. Your parcel will be off.

B--B, 2015-12-07:
Dan, Paul, I think that you' re right... That part of code has already been included but doesn' t work with the blobs that we are using... The best thing IMHO is to use the same blobs that cm use, I' ve already merged all changes made by cm in two separate branches, i' ve tried to compile but everytime i get baseband unknow, probably i forgot or made something wrong, we have a unified device tree and some things are slighty different...! Here' s the updated branches: https://github.com/AOSP-JF-MM/platform_device_samsung_jflte/tree/aosp-6.0-test-new-ril and https://github.com/AOSP-JF-MM/proprietary_vendor_samsung/tree/aosp-6.0-test-new-ril 
These branches should solve this issue... When we find what's wrong! ;)

FernBch, 2015-12-08:
7/12 test build. Caller ID is a no go.

New Logs:

https://www.dropbox.com/s/9cjt13wo1pld7ah/2015-12-07_20.19.zip?dl=0

bobwonderful, 2015-12-13:
I can confirm that this still doesn't work as of the 12/06/2015 build.

closed 2015-12-26 by B--B