AOSP-JF-MM/issue_tracking_aosp #10: Battery percentage only inside the icon.

fafualex, 2015-12-09:
Hi, I know that in cyanogenmod there's the possibility to have the battery percentage beside the icon battery and I know that rom is AOSP based , but it's really frustrating having it inside the icon because it's really small and almost invisible.
I haven't found a valid app or an xposed app for doing this like it was in the previous cyanogen based rom, so I would like to know if that is possible or not, or if you guys can put this option available in the next build.
By the way you guys are doing a really good job and now my S4 it's freaking fast!! Thank in advance.

B--B, 2015-12-27:
Unortunately, this require the revert of some aosp features, google added the system ui tuner with the battery option, must be removed before we can apply the old battery changes that we used on lp, but i love circular battery and the percentage outside the icon so... Probably the feature will be added ;)

B--B, 2016-03-20:
Finally the feature has been added 

https://github.com/AOSP-JF-MM/platform_frameworks_base/commit/a0d3ae76887b6e227215bb739f7dd7d9ebb1713b

https://github.com/AOSP-JF-MM/platform_packages_apps_Settings/commit/371f4e8b0bb6a53c80fe56db6ffff6d380a29b9f

There are some other small things to change but it's working fine, next public release will have this feature :)

closed 2016-03-20 by B--B