AOSP-JF-MM/issue_tracking_aosp #12: Touch doesnt work after Lockscreen

SaulGoodman88, 2015-12-14:
after lockscreen my touch doenst work anymore and i have to reboot the phone.
same issue with optimized cm 12.1 rom, but with other kernels the issue disappears.
iam using another cm13 build at the moment and touch works but i really like your roms.
if you need a log i can flash your rom again, the issue appears everytime.

i did a clean flash, no apps only gaaps 6.0, no xposed or other changes.

AntaresOne, 2015-12-15:
It's a known issue by the Alucard kernel as it's the only common thing across our 5.1 ROMs and AOSP 6.0. Yes, logs would be great rightly after you experience it, make sure you get (with ADB):

- adb logcat -v long > logcat.txt
- adb shell cat /proc/kmsg > /sdcard/kmsg.txt (stop this after 5 seconds by pressing CTRL+C)

SaulGoodman88, 2015-12-15:
thank you for the answer.
i tried to do a logcat, not sure if i did it right.
i can attach the logcat.txt but adb shell cat /proc/kmsg > /sdcard/kmsg.txt doesnt work it says "The system can not find the path specified ."

hope this is what you need.

[logcat.txt](https://github.com/AOSP-JF-MM/issue_tracking_aosp/files/62996/logcat.txt)

SaulGoodman88, 2015-12-15:
by the way i used TWRP v2.8.7.2 to flash the rom and did a clean flash.

SaulGoodman88, 2015-12-16:
could this be the Problem? 

[ 12-15 18:49:35.392   814:  842 I/DisplayManagerService ]

Display device changed state: "Integrierter Bildschirm", OFF

Zyphrex, 2015-12-16:
I too am having the same issue. I had the issue with 5.1 also and I believe I flashed a different kernel to the device which corrected the issue entirely. I just don't remember what kernel I actually used to fix the issue in 5.1.

It's odd that it doesn't happen when the device boots for the first time... Only when waking the device up will it become unresponsive. It's definitely a kernel issue as it happens in both system and recovery.

What alternative CM13 Rom/Kernel are you using SaulGoodman88?

SaulGoodman88, 2015-12-17:
yes it´s definitely a kernel issue. in optmized cm12.1 Rom i used the SaberMod Kernel and it fixed the issue, but i really want to use the alucard kernel because for me it´s the best kernel. battery life was awesome in the past. really hope alucard can fix this issue.

SaulGoodman88, 2015-12-23:
@AntaresOne i have an idea what could be the problem that causes this issue. After my Touchscreen was broken i changed it but the new one is an touchscreen from the deep black version (should be the I9515) and i have the I9505 Device.

Could it be that both devices use different Touch Controllers? I have read something about that flashing an I9505 kernel on an I9515 Device causes the unresponsive Touchscreen issue. Just wondering why the Stock Cm Kernel works.

hope you understand my bad english

smeroni68, 2015-12-23:
One of the differences from 9505 and 9515 is the touch... this is one of the hardware reasons why our unified roms DO NOT support 9515...

SaulGoodman88, 2015-12-23:
Didn't know that, any chance to get the Alucard Kernel working with this touch on my Device? CyanogenMod Kernel works fine. Thank you for the Answer @smeroni68.

devkid, 2015-12-26:
@SaulGoodman88 I own the S4 Black Edition (if this is what you mean with "Deep Black Edition") and it's a I9505 (at least my Status menu says so and Google results say so too). I also had the touchscreen problem on the Lollipop ROM with the Alucard kernel, but now I'm using the the official kernel provided by with the latest Lollipop ROM in the xda thread and the problem is gone.

SaulGoodman88, 2015-12-26:
@devkid yes this is the Device what i mean, i thought it was I9515. All Cm12.1 and Cm13 ROMs with Stock Kernel are working fine, i only have this issues with the Alucard Kernel. On my other Device everthing works, so it must be a problem with the Digitizer.

Would be nice if the Devs can fix this, because i really like the Alucard Kernel.

AntaresOne, 2015-12-26:
@SaulGoodman88 what has been changed there? full display assembly or just the glass? The touch screen controller is on the assembly, therefore if you change it with an I9515 one neither with stock ROMs will work... Does that happens rightly after you boot or you can unlock the lockscreen at least 1 time?

devkid, 2015-12-26:
@AntaresOne Not speaking for SaulGoodman88, but in my case the touchscreen stopped working randomly when waking up the device to the lockscreen; not every time, sometimes it worked for days straight, but eventually it would occur.

What I noticed was that there was a "gap" in the battery stats history in the time that the touchscreen was unresponsive. Before being unresponsive and after a reboot, the battery stats would be shown normally.

SaulGoodman88, 2015-12-26:
@AntaresOne  i changed full display assembly. 

It happens when i lock the screen for the first time, when i start the phone i can enter the pin and use it, but when i lock the phone the touch is unresponsive and i must hard reset the phone. 

With the August Build from optimized 12.1 Rom sometimes i can unlock the phone and sometimes not. On October Builds and AOSP 6.0 Builds i never can unlock the phone.

CidK, 2015-12-28:
I am having the exact same issue, i even tried the latest 5.1 build and am getting the same behavior. 

I have a S4 Canadian version. 

Is there anything i do to fix this ?

devkid, 2015-12-28:
@CidK Have you tried flashing the official [CM kernel](https://romhut.com/addons/cm-kernel_20151130) together with the 5.1 ROM? This fixed the problem for me.

CidK, 2015-12-28:
I will try this right away and keep you posted.

B--B, 2016-01-06:
Please try our latest release, many things has been fixed on kernel side, hopefully even this issue has been fixed...! Sorry but we cannot test by ourselves, because none of us has this kind of issue...!

SaulGoodman88, 2016-01-07:
ok i will try the latest release and report if the issue is gone

SaulGoodman88, 2016-01-07:
same problem :( with unofficial and official cm13 roms i don't have this issue.

B--B, 2016-01-10:
I've uploaded the stock kernel as addons, please give it a try and let us know if helps...!

Download link: https://romhut.com/addons/stock_kernel-183fa1da-4086-4188-b97a-10e74b38cff2

Please not that flashing this kernel will break exFAT and NTFS support so, be careful if your sdcard is exFAT or NTFS!

SaulGoodman88, 2016-01-11:
Thank you! With the stock kernel everything works fine now, no more freezes.
Would be nice if you can find out what's the problem for some devices with Alucard kernel and fix it some day. :)

micjon, 2016-01-11:
I have the same problem. If I flash ROM (AOSP 6.0) without Gapps, then touch screen works.
But if I then flash Gapps, touch screen again stops working. Touch screen is also after replacement. I have I9505.

Dionizasd, 2016-03-11:
I have the same problem, also screen was changed (i9505)
With stock kernel everything is fine. Regards

CidK, 2016-03-11:
Now am having issues with not being able to hear anything while taking or
receiving a call.

It happens every 3 or 4 calls, i need to reboot the phone in order to place
a call, and then it stop working again after 3 or 4 calls.

Its been happening on every CM13 and CM12,1

crispychiken, 2017-04-03:
I'm getting the same issue as SaulGoodman88, after the display goes off I turn it back on and touching the screen does nothing. I tested it with a lockscreen and without, getting the same results either way. 
I'm using:
Cricket Galaxy S4 SGH-I337Z
Optimized-LineageOS-14.1-20170312-Stable1
open_gapps-arm-7.1-pico-20170402
TWRP 2.8.4.0

TWRP also had this issue but I worked around it by not letting the screen turn off >_>

AthensTxMc, 2017-04-30:
I found a way around this but created another .. If you flash the stock rom it will fix the unresponsive touch screen but it breaks the bluetooth .. that is on the jdc 7.1,2 on a m919 .. I have flashed everything fresh and unless I go to addons on the OTA updates and flash the stock kernel the phone will only respond to the volume, power left. and right screen buttons but the main swipe screen is untouchable! I even turned off all security and set the swipe to none hoping that would help, tried setting a PIN and a pattern but couldn't access any of those either .. If I flash the stock kernel I get a persistent "bluetooth sharing has stopped" or something error (sorry I have been doing this for 7 hours and my wife needs her phone to work) and I have to mute that in order to do anything at all .. Anyway, in general I am surprised it works as well as it does and I appreciate all the work that went into this ..