AOSP-JF-MM/issue_tracking_aosp #13: Mobile data

sahaab, 2015-12-20:
I cant seem to connect to mobile data
Im on ufone in Pakistan 
We get 3G (H+) coverage here 
Working fine on optimized cm12.1
Using multirom, and aosp 6.0 is second rom whilr optimized cm12.1 is main

B--B, 2015-12-27:
Same behaviour with our latest release? Have you tried to see if the apn configuration for your provider is correct?

gradebg96, 2016-02-02:
Hello,

I had this problem also when I first started using the ROM.  

What helped me was: click 'Settings > More > Cellular Networks (or Mobile Network Settings) > Preferred Network Type' then change it down to 2G.  Once this switches to G or 2G, change it back to 3G or LTE.  

I had to do this the first few times when booting up (it would go for hours without picking up any mobile data unless I did this - then it worked immediately).  After doing it a few times, it seemed to always pick up mobile data immediately even after a reboot.

Hope this helps

B--B, 2016-02-02:
LTE/Data not working after a reboot has been already fixed here: https://github.com/AOSP-JF-MM/platform_device_samsung_jflte/commit/4f76a8a80292179501d4002c9a4b7c4ef1f69a09 
I close this bug, no report after our latest release.

closed 2016-02-02 by B--B