AOSP-JF-MM/issue_tracking_aosp #14: Sound issue

M3roc, 2015-12-23:
Having some issues with Spotify, the sound stutters sometimes(like when you get 100% CPU), there are also some places while playing the music, when it gets loud and then quiet again. The quality is not the same because some frequences of the music are not loud enough or played correctly.

Also wenn you put on the equalizer of Spotify the loudness of the music gets to a 100% for a very short time, i think the equalizer does also not work correctly.

SaulGoodman88, 2015-12-23:
The Audio stuttering is an issue since the latest optimized cm12.1 roms so far as I know.
Maybe they fix it in the next build.
Try to disable the AudioFX app this may fix the loudness problem.

M3roc, 2015-12-23:
Well, this would help if there even was a Audio FX app ;)

SaulGoodman88, 2015-12-24:
Haha ok sorry my fault :D

javelinanddart, 2015-12-25:
Logs please?
Dmesg and Logcat

M3roc, 2015-12-26:
Got you the logs here, did start listen to music with headphones and also turned the equalizer offline and online 2 times.

[logs.zip](https://github.com/AOSP-JF-MM/issue_tracking_aosp/files/72470/logs.zip)

B--B, 2015-12-27:
Same issue with our latest build? Spotify has been updated for MM? No issue for me listening music with google play music over bt or headphones, only some sporadic stutters when i turn of the screen, but does not happen everytime

M3roc, 2015-12-30:
It's the same issue, still some problems with random loudness in music(i think that some frequenzes are not played right and then they get loud as they should be and back to quiet) , when put equalizer on still gets lout that your ear almost will fall off for a short moment(think equalizer works fine).I also have the sporadic sutters like you have.

M3roc, 2016-01-04:
Well, since a Friend droped my phone i do not have a S4 anymore, so i will close this isse, but i still think that this is a pretty important issue, because B--B does have it as well with the sporadic stutters.

closed 2016-01-04 by M3roc

javelinanddart, 2016-01-04:
The issue is with Alucard's Kernel, if i'm not mistaken

B--B, 2016-01-04:
I was testing right now Spotify and after latest kernel changes everything it's fixed, at least for me, no more stutters and after 30 minutes of spotify usage no sound loud/quiet

ako673de, 2016-01-11:
I've been with CM12.1-Optimized together with 20151130-nightly kernel for a while (because with the latest bundled Alucard kernel I permanently ran into a Bootloop). No stuttering with that combination, ... but other issues. Finally yesterday by pure chance I found a thread called "possibly need to update modem and bootloader".  => Updated those from UHOE1 to UHOJ2 and tried again with Alucard kernel. => No more bootloops, no more "other issues" (e.g. "BT share has been stopped"). But unfortunately "audio stuttering" is back :-(. So for sure it's a Kernel issue. And it's not the only one. We have aborting WhatsApp voice messages, heavily distorted recordings, afm. I really hope that we will see one more update even though AntaresOne doesn't seem to very like to...

AntaresOne, 2016-01-11:
@ako673de http://forum.xda-developers.com/showpost.php?p=64757090&postcount=7909