AOSP-JF-MM/issue_tracking_aosp #15: Microphone not working on S4 SPH-L720

ndkar1, 2015-12-25:
Ok.  I have an S4 SPH-L720.  It was running 5.0.1 stock rooted.  I installed CWM recovery and everything worked perfectly.  I downloaded and installed the latest ROM from you guys, the correct Play Store, and the SU from you all.  Installed the ROM 1st, the store 2nd, and the SU app 3rd.  Everything works perfectly except my microphones.  No matter where I am in the ROM no microphone is ever detected.  I tried making phone calls, nothing.  I tried using Google Now, nothing.  I tried taking a video, nothing.  I'm wiping the phone and SD card and clearing the cache and dalvik cache and reinstalling the ROM now.  If I have the same issue, please tell me how to get whatever logs you may need to help me resolve the issue I'm having.  Thanks!

FernBch, 2015-12-26:
What is the name of the zip you flashed? The date of the latest release is Dec 24. If you flashed that revert back. That build is for GSM variants only. The L720 is CDMA and the build for it is not ready.

B--B, 2015-12-31:
Wrong zip, nothing to fix

closed 2015-12-31 by B--B