AOSP-JF-MM/issue_tracking_aosp #16: Bluetooth doesn't start and crashes WiFi on i337

Petelombardo, 2015-12-27:
Bluetooth on the latest 12/24 rom won't start on my i337.  Trying to start Bluetooth causes WiFi to cycle endlessly until a reboot.

Petelombardo, 2015-12-27:
12-26 21:37:20.181 E/bt_hci  (21972): command_timed_out hci layer timeout waiting for response to a command. opcode: 0xc03
12-26 21:37:20.508 E/bt_osi_config(22009): config_new unable to open file '/etc/bluetooth/ble_stack.conf': No such file or directory

smeroni68, 2015-12-27:
Just for testing purpouse, can you open with a root explorer the path /etc/bluetooth , give permissions R/W, copy and paste the file bt_stack.conf (that is in this path) with name ble_stack.conf ?
Report here what's happen. Thanks

Petelombardo, 2015-12-27:
No luck.

12-27 07:18:29.904 I/bt_vnd_conf(12503): Attempt to load conf from
/etc/bluetooth/bt_vendor.conf
12-27 07:18:29.904 I/bt_vnd_conf(12503): vnd_load_conf file
>/etc/bluetooth/bt_vendor.conf< not found
12-27 07:18:30.140 D/bt_hwcfg(12503): Chipset BCM4335A0
12-27 07:18:37.905 E/bt_hci  (12503): startup_timer_expired
12-27 07:18:37.909 E/bt_core_module(12503): module_start_up failed to start
up "hci_module"
12-27 07:18:38.294 E/bt_hci  (12503): command_timed_out hci layer timeout
waiting for response to a command. opcode: 0xc03
12-27 07:18:38.294 E/bt_hci  (12503): command_timed_out restarting the
bluetooth process.

Petelombardo, 2015-12-28:
Might this be a similar issue/resolution?
https://github.com/CyanogenMod/android_device_asus_grouper/commit/88dbab9aa8ac8590f1ff86301f569e5a62d27d55

Petelombardo, 2015-12-29:
Attaching a debug log after symlinking /etc/bluetooth/ble_stack.conf -> /etc/bluetooth/bt_stack.conf and /etc/bluetooth/bt_did.conf -> /etc/bluetooth/bt_vendor.conf to test what would happen.

[debug.zip](https://github.com/AOSP-JF-MM/issue_tracking_aosp/files/74107/debug.zip)

B--B, 2016-01-06:
Fixed here: https://github.com/AOSP-JF-MM/platform_device_samsung_jflte/commit/f1ac31ba47c48a96ae59e6da2489acddd1da0c11
And here: https://github.com/AOSP-JF-MM/platform_device_samsung_jflte/commit/f1ac31ba47c48a96ae59e6da2489acddd1da0c11

Fixes are included in our latest release, just flash the updated rom and bt will work as expected

closed 2016-01-06 by B--B

Petelombardo, 2016-01-07:
> Fixed here: AOSP-JF-MM/platform_device_samsung_jflte@f1ac31b
> And here: AOSP-JF-MM/platform_device_samsung_jflte@f1ac31b
>
> Fixes are included in our latest release, just flash the updated rom and
> bt will work as expected

I had to use adb to push the update to my phone due to the converged
storage and my lack of having previously installed supersu... BUT -- it
works!  Bluetooth is working!  Thank you! :-)