AOSP-JF-MM/issue_tracking_aosp #17: Bug installing 6.0 on my galaxy S4

directoweb, 2015-12-27:
Hello
Bug with battery and bluetooth (battery discharged in 3 hours ! and bluetooth not available ! Why ?

Here is the process I followed :
I tried yesterday evening to install the last version : AOSP-6.0-20151224-jfltegsm.zip on my i9505 S4 phone. I was already under 5.1.1 antaresone. In Teamwin recovery, I did a factory reset including wipes data, Cache and Dalvik cache.

Then installed the full range 6.0 downloaded from here, including the AOSP-6.0-20151224-jfltegsm, TWRP-2.8.7.2-F2FS, SuperSU-v2.52 and a gapps file : 
open_gapps-arm-5.1-nano-20151129
Then after install, it first appeared to work but the battery showed a file not specified precisely, using 17% of battery !
after 3 hours, my S4 was hot and closed!
Can you explain to me where  is or where are my errors ?
I have recovered on 5.1.1

B--B, 2015-12-27:
Try to clean flash two times the rom if you come from another rom, some users on xda reported that reflash again the rom helped with bt . Regarding the battery, none of us experience issue or drainage, probably something went wrong during the flash or you make something wrong.. No need to flash supersu, it' s included in our build

directoweb, 2015-12-27:
Ok thanks for information. about the gapp, wich file is the best ? >I used that one : open_gapps-arm-5.1-nano-20151129 - is ther a better version or this one fits perfectly ?
I noted also about  super su.
I will let you know.

smeroni68, 2015-12-27:
If you use the wrong gapps version, maybe you have problems!

If you open a bug report for AOSP 6.0 Marshmallow and you used a 5.1 lollipop package... come on guy!!

closed 2015-12-27 by smeroni68

directoweb, 2015-12-27:
The package was presented as a 6.0 when I downloaded it.

What is for you the right package ? where can I find it ?

Thanks for help

 

Best regards

 

 

http://www.directoweb.com/signature/logo-sign-wsi.jpg

Bruno Schouler WSI
J'optimise votre visibilité et votre réussite sur Internet
Tél : +33 9 54 47 04 05
Portable : +33 6 63 09 84 11
bruno.schouler@wsiworld.fr
Agence WSI <http://www.wsi-marketing-internet.fr/agences-wsi-francophones/agence-wsi-idf-bruno-schouler> 
Blog Directoweb <http://www.directoweb.com/> 

 <http://www.wsi-marketing-internet.fr/agences-wsi-francophones/agence-wsi-idf-bruno-schouler> certification marketing numérique WSI  <https://www.facebook.com/bschouler> Facebook  <http://twitter.com/wsiwebref> Description : Twitter   <https://www.linkedin.com/in/bschouler> compte LinkedIn de Bruno Schouler  <http://fr.viadeo.com/fr/profile/bruno.schouler> Viadeo 

http://directoweb.com/signature/Anniversary_EmailSignature.png

smeroni68, 2015-12-27:
opengapps.org

Select ARM, 6.0, and the version you like. I suggest not the smaller but for example MICRO.

directoweb, 2015-12-27:
In fact I used open_gapps-arm-6.0-mini-20151225, so it was right, no ? I just downloaded the micro : open_gapps-arm-6.0-micro-20151225.zip and I will test again on my phone.
Thanks for help received

directoweb, 2015-12-28:
I have all reinstalled on 6.0 version, quitre perfect except a new problem coming : When connecting my phone to my computer, the directories do not appear. in my computer. GT-i9505 remains empty.
If I am in recover situation, I can see the files in the phone from my computer.
Is ther something to do to repair this when phone is open ?  
On other end, bluetooth is now working, phone, sms and a lot of apps are working ok.
even the battery seems ok, but when I watch in battery, other usage (autres utilisations en français !) = 12%, ma

smeroni68, 2015-12-28:
Open the notification from top, click onn the notification of charge and select MTP...

directoweb, 2015-12-28:
Wich notification ? the top of what ? and what means MTP ? Sorry I do not understand. could you be more precise ?
Thanks

smeroni68, 2015-12-28:
This is a bug report, not a "how to use Marshmallow".... anyway, when you connect the usb cable, on the phone appear 2 notifications (on top). Slide down the notification area (i hope you know how to slide down to see the notification) and click on the one that is telling" Charge click to change " and on the list that appear select MTP file transfer. 
When you see finally on the pc your internal and external sd, please come here and close this" bug report ".... Thanks.

directoweb, 2015-12-28:
Ah okis tuhis new inandroid 6.0.1 ? I never saw this before, this status was only on on 5.1.1
Thanks

directoweb, 2015-12-29:
I discovered 2 new bugs ! the app telephone is stopping when trying to call someone ! another telephone app is working fine. and clock also is stopping when opening.
I use the "mini" gapp. I did not try with micro app. maybe different. has anyone any idea.

directoweb, 2015-12-29:
in the battery infos, the other uses is very high, without more informations, screen is 11%, wifi 10% the android platform is 4%. what do I have to do ? reinit again ? ...

directoweb, 2016-01-02:
finally, tired of getting nothing working fine on my S4, I went back to 5.1.1 optimized-CM12.1-20151130. This is fully working in all sections : phone, mail, sms, Bluetooth, 4G, and so on
Maybe later, if you succed with that rom, I will try again... ?