AOSP-JF-MM/issue_tracking_aosp #18: 19505 issues

ixkillfrenzyxi, 2015-12-28:
Sorry for not providing logs but im a bit green and dont know how. Anyway i cant turn bluetooth on, in the pull down menu tapping it wont remove the line crossing it out and in the bt settings if i move the switch to on nothing changes and the on switch is greyed out. Also a side issue is that i cant paste into the aosp browser address bar as there is no visible paste button after holding. Oh and the ota says update available but has the word null on it plus tapping it does nothing. But wanna say thanks for your work and cant wait for the teething probs to be sorted

ixkillfrenzyxi, 2015-12-28:
Oh btw this is the newest 6.0 jdct 19505gsm from xda. I used the gapps link from the same thread and installed 6.0nano. I went straight from optimised12.1 cm did a full wipe and havent messed with any kernel stuff

ixkillfrenzyxi, 2015-12-28:
Please leave this open until im able to get back on wifi tomorrow. Gonna re flash with gapps micro and see what happens since it seems that fixed somebody elses bluetooth issue

ixkillfrenzyxi, 2015-12-29:
Alas, the fresh install with micro gapps didnt work 🙁

ixkillfrenzyxi, 2015-12-29:
1. Unable to turn on bluetooth
2. Wifi signal keeps dropping and unable to reconnect unless i turn airplane mode on and off
3. Unable to past into browser address bar
4. Ota first says error reading changelog upon boot, also it says update available and null in the same box
 If you need any more info just ask

smeroni68, 2015-12-29:
Sorry, the only way is to reflash rom and gapps until it works... also 3 times or more.

ixkillfrenzyxi, 2015-12-29:
No need to apologise, I really am grateful for all the work that you do. Its just teething problems and in all honesty if it was really that bad id just revert back to lollipop. Anyways thanks for your help and i shall give it a go ✌

smeroni68, 2015-12-29:
I have a suggestion: can you change the recovery with the official twrp jfltexx and flash?

ixkillfrenzyxi, 2015-12-30:
Thanks a lot, BT issue sorted. I had flashed the rom a few times using the twrp/antares with no luck. Anyway i changed to official twrp, flashed the rom+gapps three times consecutively WITHOUT BOOTING ROM!! and voila it worked. I cant rule out the twrp i used earlier as i did boot between flashes. 
   Just so you know though, the browser that comes with the rom crashes if you try to download anything and it wont allow you to paste anything in the address bar. (Not bothered as using another browser aint no prob) 
    Matts ota app says i have an update available though trying to download it makes ota crash. 
   Um had all these other issues relating to mobile network, wifi discovery, unresponsive touch screen and unable to keep phone turned off during charging. Fingers crossed the official twrp flash ×3 trick seems to have sorted em all but not had time to mess with my phone enough to be 100%. Thanks again man

FernBch, 2015-12-30:
Go into settings>app, touch the gear in the upper right, choose app permissions, and enable storage permission for the browser.

ixkillfrenzyxi, 2015-12-30:
Haha whaddya know it works. Just no paste option, if i where to guess tho, i think its there its just hidden offscreen or behind the web page. So that app setting looks new.... Must be a marshy thing ehh. Did notice the clock was turned off but it worked this morning so darent touch that for now. Btw thanks bro

FernBch, 2015-12-30:
Much is new with MM. Now that you know how to find it you may need to enable things for some apps. It's the new permission scheme for 6.0.

ixkillfrenzyxi, 2015-12-30:
Yeah i noticed a bit of that due to installing playstore apps. Had i installed the same browser from playstore i guess  id have been asked to give it the necessary permissions as a matter of course. I do have other questions but i just wanted to post bugs here really. Thanks for resolving my issues guys keep up the good work and if i find anymore i shall let you know. For anyone with Bluetooth  issues wifi not working when cell network is low just remember flash rom and gapps , repeat another 2 times and then boot system it worked for me

B--B, 2015-12-31:
Hopefully it' s fixed after the latest changes in bt drivers in our kernel should not happen anymore

closed 2015-12-31 by B--B