AOSP-JF-MM/issue_tracking_aosp #21: Stock clock alarm inaccuracies

astiak, 2015-12-30:
Hi folks,

I'm having an issue with the stock clock app, with regards to alarms. I have set an alarm for 7am, but the next alarm shown by the clock app is completely different to what I have set.

Upon reaching the alarm time I have set, the alarm will not go off. Upon reaching the seemingly random next alarm shown by the app, there is also no alarm going off. The next alarm time does seem to extend by an arbitary amount however, each time not actually sounding an alarm.

I hope this makes sense. I have attached some screenshots to help clarify the aforementioned issues. 

I am using the latest 24/12 gsm build on jfltexx. No modifications. Using an older bootloader, but I can't imagine that causing an issue with the clock. 

Please let me know if you require any further information. 

[image]
[image]

astiak, 2015-12-31:
Interestingly enough, today the alarm did go off at the right time, despite showing the wrong "next alarm" time. I will monitor this to see if the 2 occasions where alarms did not go off may have been caused by something else. 

It may just end up being a cosmetic issue!

B--B, 2015-12-31:
Try to clean clock app data and set again your alarms, my alarms are fine.
And please report all the requested information, some gapps packages replace clock app

astiak, 2015-12-31:
Hi, thanks for the reply.

I've tried to clean data already. 

Here's the other bits of info:

Model: jfltexx 
Baseband: I9505XXUGNI2
Rome version: 20151224
Gapps: Open GApps  nano 20151225
Clean install
Rebooted: Yes
No mods
Stock kernel 
No Xposed 

I've also noticed the clock app crashes when I press the world time button. Not sure if related.

B--B, 2015-12-31:
Confirmed the crash on world time

12-31 14:37:10.024 25008 25008 D AndroidRuntime: Shutting down VM
12-31 14:37:10.025 25008 25008 E AndroidRuntime: FATAL EXCEPTION: main
12-31 14:37:10.025 25008 25008 E AndroidRuntime: Process: com.android.deskclock, PID: 25008
12-31 14:37:10.025 25008 25008 E AndroidRuntime: java.lang.NullPointerException: Attempt to invoke virtual method 'void android.support.v7.widget.SearchView.setImeOptions(int)' on a null object reference
12-31 14:37:10.025 25008 25008 E AndroidRuntime:                                                                                                                                                                                            at com.android.deskclock.worldclock.CitiesActivity.onCreateOptionsMenu(CitiesActivity.java:563)
12-31 14:37:10.025 25008 25008 E AndroidRuntime:                                                                                                                                                                                            at android.app.Activity.onCreatePanelMenu(Activity.java:2852)
12-31 14:37:10.025 25008 25008 E AndroidRuntime:                                                                                                                                                                                            at android.support.v4.app.FragmentActivity.onCreatePanelMenu(FragmentActivity.java:262)
12-31 14:37:10.025 25008 25008 E AndroidRuntime:                                                                                                                                                                                            at android.support.v7.internal.view.WindowCallbackWrapper.onCreatePanelMenu(WindowCallbackWrapper.java:85)
12-31 14:37:10.025 25008 25008 E AndroidRuntime:                                                                                                                                                                                            at android.support.v7.app.AppCompatDelegateImplBase$AppCompatWindowCallbackBase.onCreatePanelMenu(AppCompatDelegateImplBase.java:241)
12-31 14:37:10.025 25008 25008 E AndroidRuntime:                                                                                                                                                                                            at android.support.v7.app.AppCompatDelegateImplV7.preparePanel(AppCompatDelegateImplV7.java:1247)
12-31 14:37:10.025 25008 25008 E AndroidRuntime:                                                                                                                                                                                            at android.support.v7.app.AppCompatDelegateImplV7.doInvalidatePanelMenu(AppCompatDelegateImplV7.java:1527)
12-31 14:37:10.025 25008 25008 E AndroidRuntime:                                                                                                                                                                                            at android.support.v7.app.AppCompatDelegateImplV7.-wrap6(AppCompatDelegateImplV7.java)
12-31 14:37:10.025 25008 25008 E AndroidRuntime:                                                                                                                                                                                            at android.support.v7.app.AppCompatDelegateImplV7$1.run(AppCompatDelegateImplV7.java:128)
12-31 14:37:10.025 25008 25008 E AndroidRuntime:                                                                                                                                                                                            at android.os.Handler.handleCallback(Handler.java:739)
12-31 14:37:10.025 25008 25008 E AndroidRuntime:                                                                                                                                                                                            at android.os.Handler.dispatchMessage(Handler.java:95)
12-31 14:37:10.025 25008 25008 E AndroidRuntime:                                                                                                                                                                                            at android.os.Looper.loop(Looper.java:148)
12-31 14:37:10.025 25008 25008 E AndroidRuntime:                                                                                                                                                                                            at android.app.ActivityThread.main(ActivityThread.java:5434)
12-31 14:37:10.025 25008 25008 E AndroidRuntime:                                                                                                                                                                                            at java.lang.reflect.Method.invoke(Native Method)
12-31 14:37:10.025 25008 25008 E AndroidRuntime:                                                                                                                                                                                            at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:726)
12-31 14:37:10.025 25008 25008 E AndroidRuntime:                                                                                                                                                                                            at com.android.internal.os.ZygoteInit.ma [... cut ...]
12-31 14:37:10.119  3276  3276 D BatteryService: Level: 87
12-31 14:37:10.530   824   844 W ActivityManager: Activity pause timeout for ActivityRecord{743172a u0 com.android.deskclock/.worldclock.CitiesActivity t365 f}
12-31 14:37:11.418  2224  2491 W OpenGLRenderer: Incorrectly called buildLayer on View: ShortcutAndWidgetContainer, destroying layer...

I' ll take a look in next days, but don't think it's related to your issue, my next alarm time is fine... Have you tried to clear app data and cache and see if happens again?

astiak, 2015-12-31:
Hi, yep, I have cleared app data and cache. Even clean flashed again, the issue still persists. 

I'm stumped as to what else I can try! If you're not having these issues then it must be something on my end, perhaps an app interfering, but I don't use any other clock apps. I'll probe my device set up further. Since I've also not seen any forum posts on this issue it seems isolated. 

I'll post an update here if anything changes.

smeroni68, 2015-12-31:
Do not loose time... dev found the fix for it... next release will fix... thanks for head up.

astiak, 2015-12-31:
Cool, thanks for the update! Feel free to close. Happy new year gents.

B--B, 2016-01-04:
Fixed

closed 2016-01-04 by B--B