AOSP-JF-MM/issue_tracking_aosp #22: Formating external SD-Card as internal storage fails

keydon, 2016-01-02:
Hey there,

i am running your latest ASOP 6.0.1 Build from 24.Dec 15. (Clean fresh install)
But i have some trouble using the new marshmallow feature to format the external sd-card as internal storage.

### My Steps:
1. _Settings_ -> _Storage & USB_ -> Click on the _SD-Card_ -> Context-Menu _"Format as internal Storage"_.
2. The formating-wizard starts, i can choose between two options: _"As mobile storage"_ and _"As internal storage"_.
3. I select the later and confirm with _"next"_.
 The next step informs me about that the SD Card has to be formated and all data will be wiped.
4. I confirm with _"delete and format"_.
 Then the formatingprocess starts and it reaches ~60%.
5. A modal dialog appears which says: _"Your SD Card seems slow. But you still can continue this process. Apps might be slower. For better performance purches a faster card"_
6. When i confirm with _"OK"_ its does no seem to have reached 100%. Instead it jumps to the next step.
Now it calculates how much data should be moved (~4,35 GB) and is giving me two options: _Move it now_ or _later_.
7. If i choose _"now"_, its shows another screen which estimates the moving should take about 7min. When confirming that with _"move"_ the app _"Settings"_ crashes immediately.
8. If i choose _"later"_ i return to the _"Storage & USB"_-Settings page. The SD-Card now is listed as **"Damaged"** And i can start the process all over again with the same result.

Formating it as normal mobile storage works fine. Multiple reboots did not help.

I translated the labels and texts from german, so the original english wording might differ from mine described here.

I also attatched adb-logcat-logs. My first impression is that mkfs.f2fs fails for some reason.
[android-logcat.txt](https://github.com/AOSP-JF-MM/issue_tracking_aosp/files/76107/android-logcat.txt)

Thanks for your help and your great work!

keydon, 2016-01-02:
i niw also attatch the tombstone if that is any good ;o
[tombstone_02.txt](https://github.com/AOSP-JF-MM/issue_tracking_aosp/files/76113/tombstone_02.txt)

keydon, 2016-01-02:
Phone model: _GT-I9505_
Baseband version: _I9505XXUHOF2_
ROM version: _20151224_
GApps Package: _open_gapps-arm-6.0-stock-20151225_

Did you do a clean install for the currently installed build? _yes_

Did you reboot after having the issue? _yes_
Did you do any system modification? _no, besides using f2fs_
Did you flash another kernel? _no_
Did you install Xposed framework? _no_

SolGarlic, 2016-01-04:
I have the exact same problem with a standard Transcend 32GB card.

B--B, 2016-01-04:
Please read the rom discussion on xda, TWRP does not support adopted storage at the moment, fix this option actually is a nonsense, if you succesfully format your sdcard as internal storage you won't be able to update anymore your phone, and if you update the phone it will be stuck in a bootloop with the adopted storage corrupted and all the data lost.
I close this bug until TWRP guys fix their recovery, without recovery support this option makes more damages than benefits

closed 2016-01-04 by B--B

B--B, 2016-01-05:
Fixed now on rom side, but rom zip and gapps must be moved to /data/media folder (the real internal memory path), TWRP mounts the real internal memory as internal storage even when user use the adopted storage, zip can be normally selected and flashed, no issue after a couple of flashes, i will try the feature some days, if cause some kind of issue will be removed

keydon, 2016-01-06:
perfect, thank you!