AOSP-JF-MM/issue_tracking_aosp #23: Do Not Disturb Issue

FernBch, 2016-01-03:
The Do Not Disturb feature is not working properly on the 2/1 test build.

It is set to only allow calls to come through and for alarms to sound from 10pm to 7am.

Last night it did not turn on automatically at the start time (a notification alert came through at 11:15pm). Using the pull down I manually turned it on with the quick setting tile and chose for it to end 8 hours later. Do Not Disturb did end when the 8 hour window expired.

I did capture logs with syslog.

FernBch, 2016-01-03:
Logs:

https://www.dropbox.com/s/m5aby3jbgip46h3/2016-01-02_23.17.zip?dl=0

B--B, 2016-01-04:
I use the same feature and works fine for me, tested these 2 days and no issue, have you tried to delete and set again your settings?

FernBch, 2016-01-06:
I have done everything except physically deleting the affected rule. I have found that I only have this issue if I dirty flash a newer build. When I clean flash (factory reset in TWRP) the rule is present after initial boot and works as it should.

I am closing this issue as I no longer have the problem.

Thanks for looking.

closed 2016-01-06 by FernBch