AOSP-JF-MM/issue_tracking_aosp #24: Phone unresponsive during and after call

SolGarlic, 2016-01-04:
When during a call, the phone is unresponsivo: the proximity sensor or the home/power button don't turn on the screen.
Another simultaneous incomming call also doesn't turn the screen on and I cannot see who is calling me.
After the call ends I hear 3 long tones and after a few seconds I am able to turn on the phone, but only with a long press of the power button (menu button doesn't work).

You can find attached all my logs taken during a call started at 14:36 that lasted 10 minutes, with another incoming call that I did not answer.
I then turned on the screen with the above mentioned difficulties and created the log files.

[dmesg.log.txt](https://github.com/AOSP-JF-MM/issue_tracking_aosp/files/77318/dmesg.log.txt)
[event.log.txt](https://github.com/AOSP-JF-MM/issue_tracking_aosp/files/77320/event.log.txt)
[last_kmsg.log.txt](https://github.com/AOSP-JF-MM/issue_tracking_aosp/files/77317/last_kmsg.log.txt)
[logcat.log.txt](https://github.com/AOSP-JF-MM/issue_tracking_aosp/files/77319/logcat.log.txt)
[modem.log.txt](https://github.com/AOSP-JF-MM/issue_tracking_aosp/files/77321/modem.log.txt)

Model: jfltexx 
Baseband: I9505XXUHOK1
ROM version: 20151224
Gapps: Aroma GApps custom 20151225
Clean install
Rebooted: Yes
No mods
Stock kernel 
No Xposed

SolGarlic, 2016-01-04:
Also happens with baseband I9505XXUHOJ2
Might also happen with previous ROM version and with CM13

smeroni68, 2016-01-04:
Hi,
can you read this post on xda and tell us if is the same issue?

http://forum.xda-developers.com/showpost.php?p=64591021&postcount=1336

Because the answer you can find on some posts after this one... thanks...

smeroni68, 2016-01-04:
Here the important posts:

http://forum.xda-developers.com/showpost.php?p=64596511&postcount=1351

http://forum.xda-developers.com/showpost.php?p=64596850&postcount=1355

and the result:

http://forum.xda-developers.com/showpost.php?p=64609953&postcount=1376

... hope you can solve in the best way as the user confirmed!!

Regards

PS: After close the "issue"...

SolGarlic, 2016-01-04:
Thanks.
Tried searching with "unresponsive" or freeze, but didn't find that post.

closed 2016-01-04 by SolGarlic

plrodrigo, 2016-02-02:
http://repo.xposed.info/module/com.mrchandler.disableprox install this module xposed

plrodrigo, 2016-02-02:
I advise Go to setting>accesibility and power on power button hangs

plrodrigo, 2016-02-02:
I recommend you Go to setting>accesibility and power on power button hangs sorry for bad english