AOSP-JF-MM/issue_tracking_aosp #25: No signal in I9505

chusopr, 2016-01-10:
I was using 20151224-jfltegsm with no problem, but since I updated to 20160106-jflte I no longer can get mobile signal from my carrier (Vodafone Spain), not even to make calls. Surprisingly, it's able to receive SMS. This is the same that happens when I install CM 13 or other Marshmallow ROMs I found, this one was the only one that worked fine to me until its latest version.

Doing a carrier search throws an error after searching for a while.

I have made cache and data wipe. I have even formatted data and cache partitions, but only reverting to 20151224 fixed it.

Phone model: jfltexx or GT-I9505
ROM version: 20160106
GApps Package: Open GApps Nano 20160109

AntaresOne, 2016-01-11:
Baseband version? Radio logcats? (adb logcat -b radio > radio.log)

chusopr, 2016-01-13:
Baseband version: I9505XXUBMEA
Radio log: [radio.txt](https://github.com/AOSP-JF-MM/issue_tracking_aosp/files/89070/radio.txt)

AntaresOne, 2016-01-13:
We switched to the new Android 5.1 radio blobs, please update your baseband to a 5.0 version , that one you have is ancient:)

chusopr, 2016-01-16:
Yes!
I didn't even know what baseband is, but I learned how to update it and now I am using I9505XXUGNG2 and everything works better than ever.

closed 2016-01-16 by chusopr