AOSP-JF-MM/issue_tracking_aosp #28: Un-link notification and ringtone volume

jeverling, 2016-01-31:
I hope it's OK to ask for a feature here. I'm coming from CM where I was able to adjust Notification and Ringtone volume individually. I really miss this functionality, because I used to mute notifications when I wanted to sleep but left ringtone up in case someone wanted to call me in case of an emergency. The only possibility to do this I've found so far is an exposed module or I could set a silent notification tone every evening as a workaround, but it would be great if this would be possible natively.

FernBch, 2016-02-01:
You can also use Do Not Disturd. I use it to allow calls from anyone and alarms to be active and silence everything else.

jeverling, 2016-02-01:
Thanks a lot for this suggestion, I didn't realize you could allow calls there. That solves the emergency call use-case for regular calls, but it doesn't work for WhatsApp calls, I can only allow all notifications for WhatsApp or none. Also I really liked to set notifications at work to a very low volume and keep ringtone volume up to not disturb people when I receive messages but still hear it when somebody calls. So for me DND mode only goes half the way.

B--B, 2016-06-18:
This maybe could be interesting if we decide to release enhanced version of this rom with more feature, but not for our "stock" version IMHO

B--B, 2017-04-03:
This report belongs to MM, but next release will be the last, so i close this as MM reached the EOL, N is our stable release now, and unfortunately i have no time to keep two different ROMs alive for multiple devices, BTW with N stable and bug free (wifi hotspot will be fixed in next upcoming release) i don't see the sense to stay with an older android release and OMS working only in legacy mode, and in our N ROMs un-link notification and ringtone volume feature has been added

closed 2017-04-03 by B--B