AOSP-JF-MM/issue_tracking_aosp #29: Video Camera

josephshenouda, 2016-02-01:
Camera crashes whenever I use video camera, whether it be snapchat, or regular camera

plrodrigo, 2016-02-02:
Install this https://play.google.com/store/apps/details?id=com.maartendekkers.cyanapps and install cyanogen camera

plrodrigo, 2016-02-02:
And disable google camera

B--B, 2016-02-02:
This is not the right way to report a bug, however if you are using google camera it' s buggy and crashes when try to take a video, no issue for me using the stock camera app. If you are referring to the issue "Camera Error, Camera is unable to initialize.", it' s caused by mediaserver and libcameraservice and has been fixed here: https://github.com/AOSP-JF-MM/platform_frameworks_av/commit/e3810d5ffdc13c2704b7a7fa5e035ff9430073eb

josephshenouda, 2016-02-03:
wait, so how do i report bugs?

FernBch, 2016-02-03:
To properly report bugs you need to give a xlear description of the issue of the app that is INCLUDED in the ROM. Not issues with third party apps. A Logcat should be included along with how it can be repeated if possible.

The issue with Google camera has been well documented as users of many ROMs are experiencing it. Normally a reinstall from Google Play fixes the issue.

closed 2016-02-06 by josephshenouda