AOSP-JF-MM/issue_tracking_aosp #30: Bluetooth crashes audio - phone needs to restart

gradebg96, 2016-02-02:
Hello,

I did a clean flash for AOSP 6.0 on my S4.  Everything works 100% except my bluetooth audio.  If I play audio through the phone speaker or wired headphones, there is no problem at all.  If I connect my bluetooth (speaker or headset, I've tried both), it will play one song, then it will stop reading any further songs or audio tracks (for any audio player), and even if I disconnect the bluetooth at this point in time then nothing will play still.  I then need to restart the phone for audio to work.

Steps to recreate:
1. Connect bluetooth
2. Play a song (first one is fine)
3. Switch to the next song - boom it crashes

Logs attached, hope this helps and hopefully there's something simple and daft that I'm missing. The logs keep throwing up PowerHal touch errors so I've taken them out of the file.

[2016-02-02-21-22-23 - ex PowerHal.txt](https://github.com/AOSP-JF-MM/issue_tracking_aosp/files/113980/2016-02-02-21-22-23.-.ex.PowerHal.txt)
[device_info.txt](https://github.com/AOSP-JF-MM/issue_tracking_aosp/files/113981/device_info.txt)

B--B, 2016-02-02:
Our tester reported that bt is working and that bug is fixed after the upstream merge here: https://github.com/AOSP-JF-MM/platform_system_bt/commits/aosp-6.0-test-caf still some issues but even BLE devices globally works now. Regarding PowerHAL spam, i' ve removed it from build, now logs are clean and without all that crap!

simonvanderveldt, 2016-02-03:
@B--B Hope you don't mind, I have a related issue/question:
I experienced the same bluetooth failing issues, but only after installing Android Auto. Before that it worked fairly OK, but I only had +-1 hour to test it.

Now after having installed Android Auto if I don't plugin the USB cable from my car my bluetooth basically doesn't work, but once I plug in the USB cable from my car the phone says that bluetooth hands-free profile disconnected (not sure about if it says the same about bluetooth audio, will check tomorrow) but all bluetooth functionality actually works after that. I can skip songs no issue and have done multiple phones calls without issues as well.

Do you have any idea why/how this works? Is the Android Auto app forcing a different bluetooth stack or sth? Or is my phone actually working as it should now for some odd reason?

lord2k, 2016-02-18:
I have the same issue as gradebg96  on Feb-08 rom when using cardroid app or any other music player via bluetooth a2dp.

see line 45 - 57

[logcat.txt](https://github.com/AOSP-JF-MM/issue_tracking_aosp/files/136442/logcat.txt)

gradebg96, 2016-02-18:
My issue above also remains on the Feb 08 release. Dirty flash, didn't try clean flashing

B--B, 2016-03-05:
Not 100% sure, but probably I've fixed it, tried several time with our latest builds and never happened to me, even after a reboot BT is working fine without issue so.... Hopefully it's finally fixed :)

gradebg96, 2016-03-05:
I've found a work around in the meantime. After the phone reboots, play
song, pause song, play song, pause song, turn off Bluetooth device, disable
Bluetooth, enable Bluetooth, turn on device. After this, I can skip songs
no problem till a reboot then I need to do this again.

gradebg96, 2016-03-07:
Nevermind, the fix only works about half the time

marctravers77, 2016-03-20:
Same issues with BT (SCH-R970)

B--B, 2016-06-18:
Our testers reported that latest kernel changes and the new audio HAL fixes this issue, so this can be closed 

Related bugs:
https://github.com/AOSP-JF-MM/issue_tracking_aosp/issues/73
https://github.com/AOSP-JF-MM/issue_tracking_aosp/issues/71
https://github.com/AOSP-JF-MM/issue_tracking_aosp/issues/61

closed 2016-06-18 by B--B