AOSP-JF-MM/issue_tracking_aosp #33: MM Slow Animation Bug

FernBch, 2016-02-09:
Nova Launcher gave me a popup talking about the "Slow Animatiion Bug" in MM. Basically what happens is when you pull down the notification bar it sometimes drops down very slowly. I have seen it a couple rimes across test builds and is present on latest release.

Screenshot
[URL=http://s1075.photobucket.com/user/FernBchJeep/media/Screenshot_20160208-194658_zpsev0ko3gr.png.html][IMG]http://i1075.photobucket.com/albums/w434/FernBchJeep/Screenshot_20160208-194658_zpsev0ko3gr.png[/IMG][/URL]

Logcat
https://www.dropbox.com/s/fm0w0j8uy1e0uyo/Log_2016-02-08_19-47-34.txt?dl=0

B--B, 2016-02-09:
There's nothing related to a SystemUI slowdown here, webview is crashing with segmentation fault when is used by dolphin.. I do not see this issue with all other apps that use webview like the stock browser... Maybe it's something related to dolphin?

FernBch, 2016-02-10:
Not a browser issue. It haapens when pulling down the notification bar. Found out it's a known MM bug.

Closing Report

closed 2016-02-10 by FernBch