AOSP-JF-MM/issue_tracking_aosp #34: IR REMOTE CONTROL

francquadros, 2016-02-11:
Hello! It's not working the IR in my day installation 02.08.2016 , use the APK Intelligent Remote Control 3.4.0 on CM13 and TW is functioning normally , I tried the Peel Smart Remote , but also does not work , my S4 is the GT -I9505 , clean installation .

B--B, 2016-02-11:
Please submit the report with all the requested and needed information, otherwise we won't be able to help you... BTW I've tested the IR with my TV and anymote and works for me with both drivers without any issue.

francquadros, 2016-02-11:
I restored the data from BK Titanium , which had backups of apk 's
working in CM13 . always I made these restorations in all Rom's
already installed , always worked , I do it because it has several
parametric controls , but I'll try a clean install.

Franc Quadros

franc.quadros@gmail.com <nfe@flapa.com.br>

*Tel: +55 31 9668-6782*

*P*  *Antes* de imprimir pense em seu compromisso

com o Meio Ambiente e o *comprometimento com os custos.*

Johnnys10, 2016-02-14:
Thanks B--B for answering.

I activated and deactivated the USE PREVIOUS IR DRIVER in Toolbox  and it does not work the same way. I use the app PEEL SMART REMOTE and my TV is AOC (in JDC CM12.1 and AOSP 5.1 works normally).

What could be my problem?

Logs: https://mega.nz/#!vgUx0ZQC!eCSCbNHEOTCa_3pFEbMYjmaz7_z3XbpcTO0cEnTnHjQ

B--B, 2016-03-05:
Please try Anymote, go to settings, UNCHECK "Use Alterante Ir Commands" and try if works with Anymote... I'm actually using with my 2 LCD and it's working fine without any issue

Johnnys10, 2016-03-06:
Ok, in next build I'll try. Now, I'm on JDC AOSP 5.1.1 and works normally.

Johnnys10, 2016-03-14:
I try and it works, but not works well. Have some issues. JDC CM13 the peel smart works well. 

My baseband is VJUHOK1 (a little old), if I upgrade may work better?

Johnnys10, 2016-03-14:
Now, the anymote works well, thanks. 

I disabled the IR Driver in toolbox, and disabled the "IR alternative" in anymote and works well. Issue solved.

B--B, 2016-03-14:
Good, more users reported on xda too that AnyMote works without issues, this can be closed

closed 2016-03-14 by B--B