AOSP-JF-MM/issue_tracking_aosp #35: Issue with "Clear Cache" button when screen is in Landscape

FernBch, 2016-02-13:
When you rotate the screen to Landscape mode tge Clear Cache button changes to "change". In this mode it does nothing.

Screenshots to follow.

Logcat:
https://www.dropbox.com/s/kfpqe4crvy6v6hk/Log_2016-02-12_18-29-04.txt?dl=0

Portrait
[URL=http://s1075.photobucket.com/user/FernBchJeep/media/Screenshot_20160212-182754_zpskvnxow8q.png.html][IMG]http://i1075.photobucket.com/albums/w434/FernBchJeep/Screenshot_20160212-182754_zpskvnxow8q.png[/IMG][/URL]

Landscape
[URL=http://s1075.photobucket.com/user/FernBchJeep/media/Screenshot_20160212-182804_zpset0plypn.png.html][IMG]http://i1075.photobucket.com/albums/w434/FernBchJeep/Screenshot_20160212-182804_zpset0plypn.png[/IMG][/URL]

smeroni68, 2016-02-13:
This bug is in Google sources... is the same on CM13.  It's a general issue.

FernBch, 2016-02-13:
Ah. I"ll close and wait fot the update to come when Google fixes their sources.

closed 2016-02-13 by FernBch