AOSP-JF-MM/issue_tracking_aosp #36: 5 GHz WiFi not working

TabletGeek, 2016-02-13:
5 GHz hotspots do not apear under Settings-->Wi-Fi. This occurs even when manually selecting "5 GHz only" under Wi-Fi frequency band advanced setting menu. A similar issue was reported regarding 5 GHz not working when tethering but it appears that 5 GHz is disabled entirely.

smeroni68, 2016-02-13:
5ghz band is not supported in Hotspot. Only 2.4ghz.
Wifi setting is for wifi not for Hotspot.

B--B, 2016-02-13:
Duplicated of this: https://github.com/AOSP-JF-MM/issue_tracking_aosp/issues/26
 Can be closed as 5Ghz works, only tethering is broken

TabletGeek, 2016-02-13:
Since installing this ROM, my phone no longer finds *ANY* 5 GHz hotspots. I have not even attempted tethering. Even forcing "5 GHz only" does not allow 5 GHz hotspots to appear when searched. If 5 GHz is working, then please provide instructions for enabling.

B--B, 2016-02-13:
I can search and connect to my 5 Ghz network without issue, haven' tried to connect to other 5Ghz networks but i can find them when i perfom a scan

smeroni68, 2016-02-13:
You have an  M919... we have I9505...  one of the differences is baseband and base firmware.

Witch rom have you used before?

TabletGeek, 2016-02-13:
I've used the stock (rooted) TW ROM, Resurrection Remix 6.1.0 and CM13 ROMs (all with MDL baseband). This AOSP JF ROM solves connection issues that were present with the other MM-based ROMs but now 5 GHz is not working. After installing this ROM and having dropped calls I upgraded from MDL to OK3 baseband. This solved all call-related issues but did not solve the 5 GHz issue.

smeroni68, 2016-02-13:
Have you upgraded only modem or all stuff?

TabletGeek, 2016-02-13:
I am not sure what you mean by "all stuff" but, other than this ROM, only the modem was upgraded. I am still using MDL bootloader and I upgraded data and cache partitions to F2FS filesystem.

smeroni68, 2016-02-13:
all stuff means modem,  non-hlos and bootloader. 
second one is for wifi.

TabletGeek, 2016-02-13:
only modem.bin was updated (no update of non-hios or bootloader). I read that updating non-hios results in audio issues.

smeroni68, 2016-02-13:
Where you read? Anyway, is up to you. As I told you, non-hlos is the firmware for wifi and bt... In our OP on xda is reported that we use latest blobs of LP and is need to update firmware to latest one. 
Mainly I do not understand why you do not do a full update of the phone including bootloader, but is up to you... and you already tested what meant for calls... Close subject.

AntaresOne, 2016-02-13:
@TabletGeek the NON-HLOS purpose has never been 100% clear, HLOS regards the Qualcomm Hexagon DSP and it's suspected that in NON-HLOS there are audio firmware as well as wireless communication stuff. You should always update MODEM + NON-HLOS + bootloader together, having two as X version and the other as Y is never good when regards low-level shit

TabletGeek, 2016-02-13:
Where I read: The XDA post that provided instructions for updating MDL baseband to OK3 cautioned "do not flash non-hios if you have MDL bootloader. Otherwise you will experience audio problems." I would need to find that post again but this is to answer the question. I kept MDL bootloader because it does not require special work-arounds for root. If updating bootloader and non-hios will solve this 5 GHz issue (without creating other issues), then I will try that.

TabletGeek, 2016-02-13:
@AntaresOne thank you for the explanation. It looks like I need to try to flash non-hios and BL now. I will report back if that solves the problem (but it could be tomorrow when I have time to re-flash).

AntaresOne, 2016-02-13:
Mate you're on M919 which is factory unlocked, what rooting issues there are with it? Remember you could always flash TWRP with Odin (I have Odin packages in AndroidFileHost) and then flash SuperSU zip

TabletGeek, 2016-02-18:
**Update:** I've flashed the most-recent M919 official firmware, thus updating BL + BB to version OK3 (December 2015 release). However my M919 remains unable to connect to 5 GHz AP when using the JF AOSP ROM.
As a test, I flashed the [Synergy CM13 ROM](http://forum.xda-developers.com/galaxy-s4/i9505-develop/rom-synergy-s-cyanogenmod-13-0-jflte-t3272770). With that ROM my M919 connects to 5 GHz AP as it should. Possibly there is some unique issue with my M919 but, because 5 GHz works with other MM-based ROMs and I am on the latest official BB + BL, is it likely that the JF AOSP ROM is not fully compatible with the M919?
I would be willing to try to flash a different kernel with the JF AOSP ROM in order to help isolate the issue between ROM / kernel but I am not sure which kernel would be compatible. Would the Synergy kernel (used with CM13) be compatible with an AOSP ROM or is that just asking for trouble?

smeroni68, 2016-02-18:
If you check on romhut, in Addon section you can find stock kernel cm based... test with it...

zowpowow, 2016-02-20:
I have m919 too, with the 5ghz problem. I have tried the stock kernel on the latest build, and there's the same problem. The previous ROM that worked with 5ghz wifi on this phone was actually an older version of this ROM.

ke4clu, 2016-02-23:
FYI

I see you closed the issue on the no 5ghz wifi.  I installed this ROM yesterday (2/23/16) and have the same issue with not being able to see any 5ghz AP .   I had a problem with bluetooth as well.   Bluetooth would pair with my m919 and function as it should but,  when I paired it with my Eonon headunit, I had no phone service.   The 5ghz was not a deal breaker for me but the bluetooth issue was.

For testing I installed and am still using JFLTE-GPE-20151021.zip from http://forum.xda-developers.com/showthread.php?t=2557353.    JFLTE-GPE-20151021.zip ROM fixed both issues for me.

It was mention that the Baseband version may be the issue.    Well your Marshmallow ROM and the GPE ROM I am using now both have baseband  M919UVUFOH3.   So it seems that isn't the issue.

smeroni68, 2016-03-05:
Did you remember the date of this older version working on 5ghz?

zowpowow, 2016-03-06:
I think it was 20160106 but I'm not sure. I can flash it and try, but I don't know where to find old builds...

smeroni68, 2016-03-06:
https://romhut.com/versions/aosp-6-0-20160106-jflte-zip

Please test and report if it is working.

ke4clu, 2016-03-06:
I did a wipe and flash of AOSP-6-0-20160106-jflte.zip with Open gapps mini
this morning. I happy to say that the 5ghz works without a hitch now.
However the blue-tooth/phone problem persists. Blue tooth pairs with my
head set perfectly and plays Pandora or any other audio just fine but, when
I try a phone call with blue-tooth connected to my headset I can hear the
button push sounds but as soon as the phone radio turns on the blue-tooth
goes silent and the internal speaker turns on. No key presses are
recognized from that point forward. I can't hangup enter voice-mail PW or
anything else. After a few seconds maybe minute the GS4 reboots.

You want logs? If so tell me what log you want. This thing is full of them:)

Ke4clu

smeroni68, 2016-03-06:
Thanks for testing... now we can analize what can affect the 5Ghz wifi with the update made after that build.

smeroni68, 2016-03-06:
Just a test with latest official build:
- go with a Root Explorer app into the folder /etc/wifi/
- get R/W rights
- edit this file "nvram_net.txt"
- on line 12, there is "ccode=ALL"
- change ALL with US
- save and reboot

Check if any difference on 5GHz functionality.

Thanks for cooperation.

ke4clu, 2016-03-06:
I'll give that a try but I can live without the 5ghz.  I can't live without
hands free phone in my vehicle.

zowpowow, 2016-03-06:
It worked! Steps taken:

    adb root
    adb pull nvram_net.txt

Then I edited the ccode as you said. Next:

    adb shell mount -o rw,remount /system
    adb shell rm /etc/wifi/nvram_net.txt
    adb push nvram_net.txt /etc/wifi

ke4clu, 2016-03-06:
Worked here as well only now when I tried to install gapps so I could use
Google Now setup crashes.

TabletGeek, 2016-03-06:
Thank you for re-opening this issue. Due to a recent change of logic board on my M919, I can confirm that the lack of 5 GHz function is present on my "new" M919 as well (confirmed with BL/BB: MDL, NK2 and OK3). It is encouraging to see that a root cause is likely being identified. Due to the BT and 5 GHz issues (and having only 1 phone now) I had switched to a different ROM but, if it would help, I can provide logs and / or test again.
Question: can ccode be automatically updated based upon input received from the GPS or from local carrier (in order to accommodate travel to other countries)?

smeroni68, 2016-03-06:
Thanks for helping on test...

closed 2016-03-06 by B--B

ke4clu, 2016-03-06:
Thank you for the resolution.