AOSP-JF-MM/issue_tracking_aosp #37: Infrared not work

Johnnys10, 2016-02-14:
My infrared simply not work in JDC AOSP 6.0. Same issue in 34.

Phone model: GT-I9505
Baseband version: I9505VJUHOK1
ROM version: AOSP 6.0 (8 february)
GApps Package: open gapps pico (14 february)
Did you do a clean install for the currently installed build? Yes, data, cache and system in F2FS filesystem
Did you reboot after having the issue? Yes
Did you do any system modification? No
Did you flash another kernel? No
Did you flash another kernel? No

Johnnys10, 2016-02-14:
Did you install Xposed framework? No

B--B, 2016-02-14:
Duplicated of https://github.com/AOSP-JF-MM/issue_tracking_aosp/issues/34 . Please do not open a new report if someone else reported the same issue. However, as I previously said in the other report, my TV is working with anymote, I' ve just unchecked "Use alternative IR controls" and works for me. Post in the other bug report more info, device that you are trying to connect, IR app that you are using and a log, otherwise we can't help. Thanks

closed 2016-02-14 by B--B