AOSP-JF-MM/issue_tracking_aosp #39: Wifi direct

tyger1985, 2016-02-21:
Wifi direct doesn't work to send or receive files. I am able to connect to other device but nothing more. Tried with 2 Samsung S4, an S6 edge+ and Motorola Nexus 6 none works. I don't know what logs to attach. Seems other people have same problem ( http://forum.xda-developers.com/showpost.php?p=65432483&postcount=3253 )

Phone model: GT-I9505
•Baseband version: I9505XXUHPA1
ROM version: 20160208
•GApps Package:  A-GAPPS 6.0.1 (minimal)
•Did you do a clean install for the currently installed build? yes
•Did you reboot after having the issue? yes
•Did you do any system modification? no
•Did you flash another kernel? no
•Did you install Xposed framework? no

Thank You.

B--B, 2016-02-21:
I've tried now and I successfully transferred a file using wifi direct at 5 Mb/s, I've connected the devices and used es file explorer  for the share, everything works for me, please provide us a log or we can't help you

tyger1985, 2016-02-21:
I install Es File Explorer and not succed. I've made some screenshoots.
[image]
[image]
[image]
[image]

I tried create a network still failed.

FernBch, 2016-02-21:
Logcats are what is needed so the developers can see what is happening.

B--B, 2016-02-21:
As i previously said, it's working for me: 
[image]

[image]

[image]

Please provide us a log of the issue, because for me it's fine..

tyger1985, 2016-02-21:
Sorry, i didn't know what logs do you need. I install SysLog and i attatch the zip
https://mega.nz/#!DBlXWYbJ!AJ5Au6o8Kw6-b3nybrO7T714mNpui0-74xWhO79cscs

B--B, 2017-03-12:
Tested again, wifi direct works fine for me, this can be closed

closed 2017-03-12 by B--B