AOSP-JF-MM/issue_tracking_aosp #40: 5ghz WIFI not working M919

ke4clu, 2016-02-23:
FYI

I see you closed the issue on the no 5ghz wifi.  I installed this ROM yesterday (2/23/16) and have the same issue with not being able to see any 5ghz AP .   I had a problem with bluetooth as well.   Bluetooth would pair with my m919 and function as it should but,  when I paired it with my Eonon headunit, I had no phone service.   The 5ghz was not a deal breaker for me but the bluetooth issue was.

For testing I installed and am still using JFLTE-GPE-20151021.zip from http://forum.xda-developers.com/showthread.php?t=2557353.    JFLTE-GPE-20151021.zip ROM fix both issues for me.

It was mention that the Baseband version may be the issue.    Well  you Marshmallow ROM and the GPE ROM I am using now both have baseband V M919UVUFOH3.   So it seems that isn't the issue.

smeroni68, 2016-02-28:
You wrote you are using both roms... but are you using Multirom?

ke4clu, 2016-02-29:
No I'm not using multi Rom whatever that is..  I have wiped the system and reinstalled JFLTE-GPE-20151021.zip.   I have tried sever of the Marshmallow ROMS with the same result.  eg. CM13, Resurrection remix, VRToxin,  AOSP from Team Android.

closed 2016-03-06 by B--B