AOSP-JF-MM/issue_tracking_aosp #41: Microphone problem

pappapjotor, 2016-02-24:
Since I updated to feb 8th build, people can barely hear me.
Calls made through whatsapp, messenger etc is fine, it's only regular calls that are affected.
[2016-02-24_15.28.zip](https://github.com/AOSP-JF-MM/issue_tracking_aosp/files/144764/2016-02-24_15.28.zip)

pappapjotor, 2016-02-29:
So, now I've updated to HPA1.
When on 5.01 i tried to make a call, and it was just fine. But when 6.0.1 installed it's the same problem again.

B--B, 2016-06-18:
This can be closed, we switched to the new audio HAL and mic volumes now are good

More info here: https://github.com/AOSP-JF-MM/platform_device_samsung_jflte/commit/ae82ad571afc62a74ca0b86b7f6e3dd8b63489a0

closed 2016-06-18 by B--B