AOSP-JF-MM/issue_tracking_aosp #43: Bluetooth audio: error playing next song

paolo2331, 2016-02-26:
here is the link to the log 
[https://www.dropbox.com/s/jwuvrd622xogl4s/Log_2016-02-26_14-28-37.txt?dl=0](url)

The problem is around at 14:00 - 14:30 of 27/02

When I connect to Uconnect car kit music streaming has some problems.. if i push the button to change song, the error occurs and I can't play anymore any songs even choosing from the phone.

paolo2331, 2016-02-27:
Phone model: GT-I9505
Baseband version: ex. I9505XXUHOJ2
ROM version: ex. 20160208
GApps Package: Open GApps Nano 20160211 
Did you do a clean install for the currently installed build? yes 

Did you reboot after having the issue? no
Did you do any system modification? no
Did you flash another kernel? no
Did you install Xposed framework? no

B--B, 2016-03-05:
Duplicated of https://github.com/AOSP-JF-MM/issue_tracking_aosp/issues/30

closed 2016-03-05 by B--B