AOSP-JF-MM/issue_tracking_aosp #45: "Sim Card Removed" Must restart phone.

codeseven58, 2016-03-04:
It's been occurring for quite some time. today it happened 4 times. When I first installed the 20160208 release, it was happening up to 8 times/day. After wiping cache and reinstalling enough times something happened and it began occurring again. The most recent incident happened sometime within 10 minutes of the end of this file.
[2016-03-03-17-59-33.txt](https://github.com/AOSP-JF-MM/issue_tracking_aosp/files/158086/2016-03-03-17-59-33.txt)

codeseven58, 2016-03-04:
Performing complete wipe and starting over. Wiping Dalvic/Art Cache, Cache, Data, Internal Storage, Preload, and System.

codeseven58, 2016-03-04:
Issue persists after complete wipe.
https://drive.google.com/file/d/0B92I26nDotirUkZMMFdGa3MtRVk/view?usp=docslist_api

codeseven58, 2016-03-04:
Phone model: SGH-M919
Baseband version: m919uvufok3
ROM version: 20160208
GApps Package: opengapps pico 20160308

Did you do a clean install for the currently installed build? yes and no

Did you reboot after having the issue? yes, no choice.
Did you do any system modification? yes: x-posed with no-charging-light mod. also occured without xposed installed and a clean wipe.
Did you flash another kernel? no
Did you install Xposed framework? yes and no

javelinanddart, 2016-03-04:
Have you tried updating your modem to one that's post JB?
It could also be a hw issue

smeroni68, 2016-03-04:
I suggesto to do a clean flash of another rom that has worked for you before and check if the problem disappear... it's the best way to diagnose if is an hw problem.

codeseven58, 2016-03-05:
Not a hardware issue, not present in stock firmware.

And I haven't found another rom that is worthy of trying out :-\

However I may have found a solution. I went in and reset the APNs to default via the menu at the top right in the cellukar network zection. I havent had an issue since. Although, i just realized doing this also defaulted my APN to brightspot, which is what I've been using for the past day. Im going fo set it back to T-Mobile LTE and see if it starts to happen again.

codeseven58, 2016-03-06:
Update, after realizing I was on the brightspot APN and switching back to t-mobile LTE, i got another "sim card removed" error 24 hours later. I am now back on brightspot to see what happens.

codeseven58, 2016-03-09:
2 days since last "sim card removed" error. restting APNs to default and defaulting to brightspot seems to have fixed the issue. also wiped dalvik after swapping.

codeseven58, 2016-03-10:
3 days.... will post next reply either when i get an error or after a week

codeseven58, 2016-03-13:
more errors started sprouting for unknown reasons. resetting APNs wasn't an appropriate solution. perhaps this can point someone in the right direction?

AntaresOne, 2016-03-20:
@codeseven58 Still experiencing that issue? First make sure the SIM card is fine. If so, your SIM card module needs to be changed. It's relatively easy to do it and you could save up a lot of money by repairing yourself. Try to search "I9505 SIM card reader" on eBay or Amazon, this spare part is the same for any S4 LTE

codeseven58, 2016-03-22:
I'm sorry AntaresOne. After enough errors I followed the instructions [here ](http://forum.xda-developers.com/galaxy-s4-att/general/how-to-fix-galaxy-s4-sim-card-removed-t3113480)and have not had an error since. I guess the Issue is hardware related. More specifically, the connector doesn't make good contact either after some time or after being dropped so many times. Thank you for your time!

closed 2016-03-22 by codeseven58