AOSP-JF-MM/issue_tracking_aosp #46: Features

RobbeDelarbre, 2016-03-05:
Hello,
This rom is really good but can you add some features?
-buttons: so i can change the menu button to the recents button (I really miss this feature if this feature doesn't comme i will go back to your 5.1.1 rom)
-quick settings: can you make more quick settings because i only use 4 quick settings that are now availeble 

I don't know if i should poste tis here because this is for bugs but i didn't know where to poste it else.
Also sorry for my bad englisch i am not a native englisch speaker :)

codeseven58, 2016-03-08:
sorry bud. the devs won't be adding any features and asks everybody not to request them.

B--B, 2016-03-20:
Hi, hw buttons feature has been added (a lot of users requested it so we decided to add it), quick settings require more work and SystemUI tuner must be removed due to all the restrictions in the code, actually it' s not a feature that we are planning to add..!

closed 2016-03-20 by B--B