AOSP-JF-MM/issue_tracking_aosp #47: Blue-tooth and phone

ke4clu, 2016-03-07:
Hi, 
Thanks for all the assistance with the 5ghz problem.   I have another one that I'm perplexed about.  Between my wife and I we have three blue-tooth head-sets.    I also have a pure Android head unit in my truck.   The M-919 will pair with all three blue-tooth head-sets and play audio and make phone calls all day.  That didn't work until after we got the 5ghz problem fixed.   In the truck it will pair and play music all day long but will not make any phone calls.  The phone options are not even available ion the head unit.
I restored Lollipop GPE 5.1.1 (JFLTE-GPE-20151021.zip) from a backup and blue-tooth works everywhere just as it should including the Android head unit. 

By the way the way I try and use different ROMS is by backing up and restoring.  I tried Multi-ROM but my phone go stuck on the boot animation. 

Phone model: SCH-M919
Baseband version: M919UVUFOH3
ROM version: AOSP-6.0-20160208-jflte.zip

GApps Package: CM GApps [date] or Open GApps [type] [date] or Slim GApps [type] [date] ex.  Open GApps Mini 20160307

Did you do a clean install for the currently installed build? yes
Did you reboot after having the issue? yes
Did you do any system modification? no
Did you flash another kernel? yes: no
Did you install Xposed framework? No

KE4CLU

codeseven58, 2016-03-08:
No idea what a "pure Android head unit" is. Elaborate, ¿por favor?

ke4clu, 2016-03-08:
Thank you for the response.   
It is a double din in dash radio navigation unit that runs pure android OS.  Here is a link to the one I have.  http://eonon.com/Car-GPS/Android-Car-GPS/GA2114.html
Thanks

codeseven58, 2016-03-10:
and youve tried forgetting all bluetooth connections and reconnecting?

ke4clu, 2016-03-10:
Yes I have done that.   I also whet through system settings on the head
unit and tried every model of blue-tooth adapter available on the head-unit
MCU.   I switch back to lollipop and everything is working.

codeseven58, 2016-03-11:
hmm. include some logs of when youre trying to connect.

ke4clu, 2016-03-11:
I'm sorry but I'm not too up on what log and where it is.  Please advise
and I'll include it.

ke4clu, 2016-03-12:
One more problem has manifested itself.  I purchased a 0K-SP3 (KanKun) smart switch.  While trying to program the switch to my network it would never connect to my router.  After a day of trying to determine what I was doing wrong I did a nandroid restore back to GPE-5.1.1 from JfLTE and it programmed the first try.  I think I will stay with GPE until the bug are worked out of this version of 6.0.1.  Thanks for all your assistance.

closed 2016-03-12 by ke4clu