AOSP-JF-MM/issue_tracking_aosp #48: SuperSU 2.65 not working on SCH-I545

FernBch, 2016-03-15:
If I clean flash or dirty flash any build with SU2.65 I get a message about a missing/uninstalled SU binary.

[IMG]http://i1075.photobucket.com/albums/w434/FernBchJeep/Screenshot_20160314-190730_zpslems0ffk.png[/IMG]

If I flash SU2.52 during the flash process then update to 2.65 via Play 2.65 works as it should.

FernBch, 2016-03-15:
Forgot to mention this is from when I dirty flashed the 14/1 build of AOSP.

B--B, 2016-03-15:
If you dirty flash the rom and supersu is in data/app (like playstore updates) then you have the same supersu version  on data/app and system/app...

FernBch, 2016-03-15:
I get the same result if it's a clean flash.

FernBch, 2016-03-15:
Let me add to the above comment. If I flash the ROM clean I get the missing SU binary message after boot. I then have to reboot into TWRP and flash SU2.52 and reboot. After bootup SU works properly regargless of if it gets updated or not.

B--B, 2016-03-20:
I' ve updated supersu with latest 2.68 version, can you give it a try? And another question: If you flash only supersu 2.65 zip does it work or su is missing like when flashing the rom?

FernBch, 2016-03-21:
It will be a day or two before I gat back. I'm currently testing the 12/3 Opt CM13 for @side (have a workaround for No SIM on CDMA)

I should be goung back to AOSP on Tuesday afternoon. I'll clean flash at that time. I'll also have the diffetent versions of SU on ExtSD at that time and check everything at the same time.

FernBch, 2016-03-22:
Flashed latest test build today. SU2.68 install fails during flashing process. Any version above 2.52 gives the same error message but it will update to 2.65 via Play. I have a recovery log but when I view it with Root Explorer it says it is truncated. Message is similar to this one reported on XDA

Boot image patcher
***
-- Finding boot image
--- Boot image: /dev/block/mmcblk0p20
- Extracting ramdisk
- Decompressing ramdisk
--- Failure, aborting

https://www.dropbox.com/s/xt7c2q2wygo4wn7/recovery.log?dl=0

FernBch, 2016-03-26:
I've come to the conclusion that this issue is SU related, not ROM related. SU flash fails when it gets to the decompress ramdisk step. I had posted in the SU thread some time ago about this and have seen no resolution.

At this point in time I see no reason to keep this issue open. It only seems to affect this variant (SCH-I545). Maybe if the MDK bootloader could be inlocked it could be different.

closed 2016-03-26 by FernBch

murph17, 2017-06-08:
try SuperSU with System Mode 2.76 or 2.79
https://forum.xda-developers.com/apps/supersu/v2-64-2-66-supersu-mode-t3286120