AOSP-JF-MM/issue_tracking_aosp #49: Bluetooth Share crash

efmbrito, 2016-03-16:
Been testing this ROM since Beta 5 release and in that one and also in the latest Beta 6 had several Bluetooth crashes 
In both ROMs did a clean install and have recently flashed Bootloader/Modem HPA1 (same situation with older bootloader/modem).

The BT devices paired in this moment are:

- Moto 360
- Pioneer AVH-X8500
- LG Handsfree Speaker

 Hope it helps.

Logs attatched:
[2016-03-16_17.37.zip](https://github.com/AOSP-JF-MM/issue_tracking_aosp/files/176551/2016-03-16_17.37.zip)

smeroni68, 2016-03-16:
We do not have BT share FC anymore... In witch condition you get crashes? Play music, calls...

Can you detail your installation files and sequence (gapps and other)?

efmbrito, 2016-03-16:
Unfortunally i have. After boot the bt connection is established and after a while,  out of the blue,  it begins the BT Share FCs. Not in call or using a2dp. 
Have TWRP 3 and f2s partitions. 
Last installation wiped dalvik, cache,  data and internal storage,  installed beta 6 +  opengapps (Arm,  6.0,  nano,  v15032016),  wiped dalvik and reboot. 
 Same ocurred with beta 5 and other opengapps nano date versions.

smeroni68, 2016-03-16:
Have you try micro gapps?

efmbrito, 2016-03-16:
No i did not. Should it make a diference? Isn't bluetooth part of the rom? Sorry for asking but I'm a newbie in this matter

smeroni68, 2016-03-16:
We use for testing Micro package... please try with it and check if something change. Thanks

efmbrito, 2016-03-16:
Ok,  i will try and leave feedback. 
Thank you

efmbrito, 2016-03-17:
Hello Stefano,

 Same problem. AOSP Beta 6 and opengapps micro (v 20160317).
 Logs attatched: 
[2016-03-17_15.29.zip](https://github.com/AOSP-JF-MM/issue_tracking_aosp/files/178162/2016-03-17_15.29.zip)

Could it be something related with LE BT? I will unpair the Moto 360, reboot and verify.

efmbrito, 2016-03-17:
Steffano,

 No paired devices with phone, reboot and problem persists. After a while bluetooth share crashes.
Log: [2016-03-17_16.15.zip](https://github.com/AOSP-JF-MM/issue_tracking_aosp/files/178236/2016-03-17_16.15.zip)

 I really hope you have time to catch the problem.

Thanx

B--B, 2016-03-17:
Did you restore your old settings with Titanium backup? Unfortunately there's nothing that can help us in logs, just a timeout for the axi_bridge_lock when the bt is enabled (this is the causing your crash), and it's the first time that i see this issue..

efmbrito, 2016-03-17:
Hello,

  Thank you for the answer.
  No I did not restore the phone settings with TB. Is there any information that I can collect that might help you guys? 
  Or I should give chance to CM13 Optimized to see if the problem persists?

smeroni68, 2016-03-17:
That's a good idea... if you can test with Optimized CM13 and let us know.

marctravers77, 2016-03-20:
it doing the same thing with the BT share (SCH-R970)

efmbrito, 2016-03-21:
Hi guys,

  I've tested the Optimized CM13 during weekend and, after a day of use, the BT Share crash began. 
  Tried to test Optimized CM12 but had a few problems (after apps update and reboot had the loop of optimizing apps and reboot). Maybe I'll try this later but during this next week I'll mantain stock Lollipop.

efmbrito, 2016-03-24:
Hello,
  Until this moment, after 3 days of use of Optimized CM13 (12 March version), no BT Share crashes (only some audio stuttering in A2DP).

  Last week history:
Saturday and Sunday
- Reflash to stock LL 5.0.1 (modem/bootloader I9505XXUHOA7) through Odin. Used it for two days with no problems and no limitations;
Monday until now
- Rooted phone with CF-Auto-Root
- Installed TWRP
- Converted cache, data and system to f2fs
- Installed CM 13 Optimized (12 March)
- Installed OpenGapps Micro (ARM, 6.0, Micro)
- Did not paired Smartwatch Moto 360 (did not installed Android Wear neither Motorola Connect)
- I have been using BT with Pioneer AVH-X8500 (Car AV System), Peugeot SMEG+, LG Handsfree Speaker, and Pioneer HTZ-606BD (Home Cinema). As stated before only some audio stuttering with A2DP.
 
  Next steps:
- Nandroid Backup
- Install Android Wear and Motorola Connect
- Pair Moto 360 (LE BT 4) with the phone 

Will post results soon

efmbrito, 2016-03-24:
Hi again, 

 I've installed only Android Wear and paired Moto 360. After a couple of hours BT Share FCs started to appear. 
  Will try later to flash stock kernel to see if it makes any difference.

efmbrito, 2016-03-25:
Hello,

Uninstalling Android Wear and unpairing Moto 360 did not resolve issue, neither wiping Dalvik and Cache. I had to restore nandroid backup.
Regarding stock kernel, where can i find it? Stock kernel that you guys have in romhut in AOSP 6 does not works with Optimized CM13.
Thanx

simonvanderveldt, 2016-04-02:
I'm having bluetooth share force closing issues as well. Had it with the 20151224 release maybe once or twice a day, with the 20160314 release I have them all the time, most of the time even multiple overlapping force close pop-ups.

They don't happen automatically after a reboot, but do happen after I've connected to my headunit (VW golf 2016). What can/should I provide to debug this? A complete logcat?

rdchase, 2016-05-17:
I have the same problem on the April 10th version of the aosp 6.0. I didn't have any until about 8 hours after pairing my smart watch(moto360).

B--B, 2017-03-12:
No one reported Bt share crash in our latest releases, i close this report as a lot of time has passed, if happens again open a new report with logs, thanks

closed 2017-03-12 by B--B