AOSP-JF-MM/issue_tracking_aosp #50: Mounting issues with External SD card; logcat issues (related?)

logomancer, 2016-03-18:
The 20160314 build seems to have caused an issue with mounting SD cards. My ext4 SD card won't mount; the notifications say it's corrupt, though it reads fine elsewhere. Attempts to mount fat32, ext2, and exFAT formatted SD cards succeeded in mounting, but read-only. Logcat from one boot cycle (where I inserted and removed the SD card a couple of times) suggests that vold is mounting it incorrectly; see line 3166 of [this gist](https://gist.github.com/logomancer/1169e0f5e73532977fe2). Logcat is only partial, however, since logcat output appears to crash several seconds after booting. Subsequent attempts to read the logcat produce the "logcat read error" error.

Phone Model: SGH-M919
ROM Version: 20160314
GApps Package: OpenGApps Nano 
Baseband Version: M919UVUFOK3

Did you do a clean install for the currently installed build? No, this was a dirty install from 20160208. Cache/Dalvik was cleared between builds.
Did you reboot after having the issue? yes
Did you do any system modification? no
Did you flash another kernel? no
Did you install Xposed framework? no

logomancer, 2016-03-18:
External SD card ext4 mounting issue confirmed on clean install (wiped system, data, cache, dalvik). Tried another SD card with an exFAT partition and it seemed to work after designating it as external storage. Not sure what's going on there.

B--B, 2016-03-18:
Hi, thanks for the report, it's fixed now...! Btw logcat works fine, just enable logcat on STweaks (Other Tab) and will work without any issue.

closed 2016-03-18 by B--B