AOSP-JF-MM/issue_tracking_aosp #53: NOT AN ISSUE.JUST QUESTION

dkati, 2016-03-22:
hello :)
any ideas of why cant switch network type ?

thats from adb

NetworkSettings: handleSetPreferredNetworkTypeResponse: exception in setting network mode.

thanks

B--B, 2016-03-27:
Hi, sorry for being late but i'm really busy in these days...! But are you referring to our cm? In that case i've fixed the network switch yesterday, see here: https://github.com/JDCTeam/android_frameworks_opt_telephony/commit/eb30f80d249d8df9704d2593f7406f075e7354c7
I've merged all latest cm changes, solved all conflicts and fixed upstream merge option in our script, now we can use the build script for merge all the code without the need of picking a single commit everytime..!

dkati, 2016-03-27:
thanks for that :/ 
was too ''deep'' for me..

B--B, 2016-05-05:
I think that i can close this now..! :-)

closed 2016-05-05 by B--B