AOSP-JF-MM/issue_tracking_aosp #58: No APN issue

marctravers77, 2016-04-18:
This has been an issue in previous MM builds.... There isn't an option to configure apn like in 12.1 version.    Also,  when using an app to finally get to edit the APN settings.   It says that configuration for this user is not available.   I'm using a SCH-R970 on USCC

FernBch, 2016-04-18:
If you go back to the previous release of this ROM do the APN settings become visible again? A)so, do you have any kind of service?

marctravers77, 2016-04-18:
I do  know the last AOSP build  was the same, and  on the Optimized CM13  the  most recent build on that too.   The services that do work are LTE,  SMS,  PHONE.  NO MMS though.  There isn't any "Access Points"  option in Mobile networks to edit APNs.   Also,  when I do use a apn app to get to where I can edit the APN settings it says  something like "Access Point settings cannot be edited by this user".  I'm currently on Optimized CM 12.1  January build with no issues all services work fine and I can access APNs settings.

Baseband: R970TYUGOH2

Carrier : USCC

MODEL: SCH-R970

https://github.com/AOSP-JF-MM/issue_tracking_aosp/issues/58#issuecomment-211348428

FernBch, 2016-04-18:
Ok. You do not have to do this, but.......

I had a very similar issue on the I545 for Verizon recently with the exact same symptoms. What I discovered was there was some kind of issue or corruption of the EFS folder. I flashed a stock ROM via Odin, used the IMEI Backup tool created by Antaresone and made a complete IMEI backup, flashed the ROM I was having issues with, then restored the IMEI backup. I also had to dial *#*#4636#*#*, choose "Phone information", touch the menu on the upper right, and choose "Enable Mobile Data". Once I did those things LTE data began to work again.

May work for you, too. You can try dialing #*#*4636*#*# first and enable mobioe data to see if that fixes your issue 

Hope it helps.

FernBch, 2016-04-18:
Just incase the dialer code does not show properly omit the spaces from the following:

* # * # 4636 # * # *

marctravers77, 2016-04-19:
LTE and SMS works just fine... It's MMS that isn't.. Will there be a fix soon in the build for this?

https://github.com/AOSP-JF-MM/issue_tracking_aosp/issues/58#issuecomment-211476024

B--B, 2017-04-03:
This issue belongs to MM, but next release will be the last, so i close this as MM reached the EOL, N is our stable release now, and unfortunately i have no time to keep two different ROMs alive for multiple devices, BTW with N stable and bug free (wifi hotspot will be fixed in next upcoming release) i don't see the sense to stay with an older android release and OMS working only in legacy mode

closed 2017-04-03 by B--B