AOSP-JF-MM/issue_tracking_aosp #60: Unknown number on outgoing calls

Jazas, 2016-04-22:
Call log doesn't show numbers on outgoing calls, they appear as "Unknown". The interesting thing is that if I dial the number, but hang up before someone answers it, number is shown correctly, but if I call, make a conversation and then hang up, then call log doesn't show the number which I just called (it doesn't matter if I dial the number manually or call from my contact list). 
Another thing I should mention is that this ROM's 12.24 build hadn't got this issue, but after 01.06 build it appeared and exists since. 

This problem is keeping me from using MM ROM, I was using Lollipop ROM instead, because JDC's AOSP Lollipop ROM doesn't have this issue.

I guess you understood my issue description. I really want to help you to solve this problem, please don't hesitate you ask me for more information, I will try my best. Keep up the good work, you're awesome!
 
    Phone model: GT-I9505
    Baseband version: I9505XXUHOL2
    ROM version: ex. 20160416
    GApps Package: Open GApps Aroma 20160420

    Did you do a clean install for the currently installed build? yes
    Did you reboot after having the issue? yes
    Did you do any system modification? yes: no
    Did you flash another kernel? no
    Did you install Xposed framework? no
[calltest.txt](https://github.com/AOSP-JF-MM/issue_tracking_aosp/files/232147/calltest.txt)

B--B, 2016-04-22:
Should already be fixed, have you enabled the "Connected line identification" option under Settings-->Other-->Cellular networks?

BTW this issue is not rom related, but belongs to your provider, see more here: https://forums.lenovo.com/t5/MOTO-G-MOTO-G-4G-LTE-1st-Gen/Problems-after-update-to-Lollipop-call-history/m-p/3233050#M22451

Jazas, 2016-04-22:
Thank you for reply. I tried toggling "Connected line identification", but nothing has changed. Tomorrow during working hours I will call my service provider, maybe they will help somehow. 

One thing I still don't understand. If this issue belongs to my service provider, why there was no issues with JDC's AOSP Lollipop ROM? What was changed from that version?

B--B, 2016-04-22:
Ril blobs has been updated to lp tw version, we were using kk blobs from gpe in our lp rom..! 

https://github.com/AOSP-JF-MM/proprietary_vendor_samsung/commit/919b4e78733be2ab316c6e23600016969ae7655a

Please post me the full log of the issue, I will take a look at colp code to see if something is wrong...!

Jazas, 2016-04-22:
I included log in the OP. Should I post anything else?

B--B, 2016-04-22:
Ops I saw it only now...! I will take a look at it tomorrow and see if I need something else..!

Jazas, 2016-04-26:
Update: I contacted my service provider and they said that they will inform their smartphone experts about my problem. After few days I noticed that the problem was solved, just like magic. I didn't understood why suddenly everything was fixed. So I wrote again to provider and asked what they did and they said nothing... Can anybody explain what happened? Now I'm really happy and confused at the same time.

closed 2016-04-27 by Jazas

B--B, 2016-05-05:
Hi, sorry for being so late, unfortunately real life come first... Glad that works, even if honestly i don' t have  a good explanation...! :-) I took a look at colp code and it's the same as cm, everything should workas expected. i can't test by myself because my provider does not have this issue, but i don't see any other reason that could cause this problem...!

timjosten, 2018-06-24:
Basically, there is a bug in Android since Lollipop, which causes such behavior. It can be easily fixed with this patch:
`diff --git a/src/java/com/android/internal/telephony/Connection.java b/src/java/com/android/internal/telephony/Connection.java
index 21b116d..a6cb6bf 100755
--- a/src/java/com/android/internal/telephony/Connection.java
+++ b/src/java/com/android/internal/telephony/Connection.java
@@ -241,7 +241,7 @@ public abstract class Connection {
      */
 
     public String getAddress() {
-        return mAddress;
+        return (mAddress == null || mAddress.isEmpty()) ? getOrigDialString() : mAddress;
     }
 
     /**`
I have also created Xposed module to fix this problem on closed source ROMs:
http://repo.xposed.info/module/ru.timjosten.unknownnumberfix