AOSP-JF-MM/issue_tracking_aosp #61: Marshallow phone over bluetooth on car radio

ke4clu, 2016-04-25:
Phone model: jfltmo SGH-M919
    Baseband version: M919UVUF0H3 
    ROM version: 20160204

    GApps Package: Open GApps Full

    Did you do a clean install for the currently installed build? yes ( wiped with TWRP )
    Did you reboot after having the issue? yes
    Did you do any system modification? yes: no
    Did you flash another kernel? yes: stock kernel available on RomHut
    Did you install Xposed framework? no

I though I'd give this ROM another go.   Phone over BT works perfectly on three different BT headset my wife and I have.   But, in the car BT pairs 1st try every time.  Music from Pandora or MP# files from my phone play perfectly over the car audio system.   The only think is that when I pair the phone with the car it says  Connected, Connected (no phone), Connected, Connected (no phone) like 4 time before it settles on Connected (no phone).   I can make calls directly from the phone and it mutes the BT audio as long as the call is active.  Hands free driving is an impossibility at the moment.  I have cleared the Bluetooth cache on the car radio but see know way to do so on the phone.  I have deleted all Bluetooth pairings and repaired.  Nothing helps.  
I did a google search  ( marshmallow Bluetooth problems with car phone )  and get many hits.  It seems to be a major problem with even factory releases. 

Any suggestions?
Thanks

andreiciocanvlad, 2016-05-03:
i had a similar problem, when sharing music over bluetooth in my car, connection drops or music starts to glitch or the music app is unresponsive and skips tracks. Last built i hade the FC of bluetooth share.

ke4clu, 2016-05-04:
Funny thing.  A flashed back to 5.1.1 build LMY49J and everything works fine.  Bluetooth audio and phone over Bluetooth on the car audio system and Bluetooth headsets.

B--B, 2016-05-05:
Hi, please wait the latest public build and let us know if works. i've completely reworked the bt repos, now code is synced with cm, if works on our opt-cm-13.0 then should work fine here too!

ke4clu, 2016-05-05:
B--B,  If I am understanding you correctly you would like for me to install the latest build and see if it works.  AOSP-6.0-20160416-jflte.zip is the latest build on Romhut and that is the build I have installed.  in the mean time I'm going to download the cm-13 nightly, cm-13.0-20160505-NIGHTLY-jfltetmo.zip, and see how that works.  I have tried cm13 builds in the past and the phone would lockup when every I tried to use the phone over my headset.  The only way to get it to stop was wait for a timeout or manually turn Bluetooth off.  I'll let you know how it goes here on this thread. .

ke4clu, 2016-05-05:
B--B,  I flashed cm-13.0-20160505-NIGHTLY-jfltetmo.zip and found Bluetooth dysfunctional.   Well, not completely.  The Bluetooth will connect to my headset but when i try to make a call I hear the tones from the dialing but as soon as the call is started, Bluetooth sound goes dead and the keypad on the phone does not function anymore.  The only way out is the wait for a time out and turn Bluetooth off or hold power until the phone shuts off. .

B--B, 2016-06-18:
Our testers reported that latest kernel changes and the new audio HAL fixes this issue, so this can be closed 

Related bugs:
https://github.com/AOSP-JF-MM/issue_tracking_aosp/issues/73
https://github.com/AOSP-JF-MM/issue_tracking_aosp/issues/30
https://github.com/AOSP-JF-MM/issue_tracking_aosp/issues/71

closed 2016-06-18 by B--B

ke4clu, 2016-06-28:
I just installed the latest release dated 20160612.  That cured the problem with not being able to use hands free calling in the car.  It also introduced another issue with my phone.

B--B, 2016-06-28:
Latest release is 20160619 https://www.androidfilehost.com/?w=files&flid=70551

ke4clu, 2016-06-30:
OK, so I installed 20160619 fresh install.   Radio issue was resolved but I
get the infamous Unfortunately, Bluetooth share has stopped error randomly
and frequently  so the ROM is unusable for me.

B--B, 2016-06-30:
As far as I can see no one reported multiple bt share crash, and no one reported any useful information, someone stated that the crash is caused by latest gapps, but no one provided logs or other info, I barely use bt but I never had bt share crashes

ke4clu, 2016-07-01:
I'm not sure what logs you would like to see but if you will give me the file names I will be happy to reinstall and send them to you.