AOSP-JF-MM/issue_tracking_aosp #64: Bluetooth 4.0 LE not working

bestsase, 2016-05-05:
I am not able to pair and connect with my Mi Band 1s...
than noticed that i can't connect to my Nexus 7 either, it Pairs with it but not connecting...

B--B, 2016-05-05:
Latest public build? In that case wait the latest public build and let us know if works. i've completely reworked the bt repos, now code is synced with cm, if works on our opt-cm-13.0 then should work fine here too!

bestsase, 2016-05-05:
20160416 version, so ye probably its latest.. Also yes in cm13 it works okay, but not on aosp :(

Imgiling, 2016-05-07:
I confirm that I also can't pair the phone with my Xiaomi Band 1S using Stable 1 from 2016/04/16.

JoyMachine, 2016-05-13:
Same here... I have tried to connect Polar H7. It just trying to pair and after 30s it stops.
On CM-13-20160510 it connects in a few seconds.

B--B, 2016-06-18:
I close this bug cause i synced my bt branches with cm, now everything works like in cm and people reported on xda that this issue is gone

closed 2016-06-18 by B--B