AOSP-JF-MM/issue_tracking_aosp #67: No sound during calls

Jazas, 2016-05-12:
Almost every call that I answer has no sound. Neither I get any sound, nor other side can hear me. There is no sound also when I make the call. Rebooting seems to help just for 1 or 2 calls. Also, I found out that disabling and re-enabling dialer app sometimes solves the issue. Usually I reboot my phone ~10 times a day, it gets really annoying.

I included logs where I can identify few errors associated with audio, but I'm not an expert.
[Log report.txt](https://github.com/AOSP-JF-MM/issue_tracking_aosp/files/260761/Log.report.txt)
 
Any help to solve the problem would be appreciated. If I can't get solution for this bug I have to go back to Lollipop ROM again, which was the most stable so far.

Phone model: GT-I9505
Baseband version: I9505XXUHPA1
ROM version: ex. 20160416
GApps Package: Open GApps Aroma 20160420

Did you do a clean install for the currently installed build? yes
Did you reboot after having the issue? yes
Did you do any system modification? no
Did you flash another kernel? no
Did you install Xposed framework? no

B--B, 2016-06-18:
This https://github.com/AOSP-JF-MM/platform_device_samsung_jflte/commit/ae82ad571afc62a74ca0b86b7f6e3dd8b63489a0 shuld fix the issue, but can't confirm because this bug never happened to me

dookiedude, 2016-06-19:
This issue occurs sometimes when you have Google Now from any screen enabled. The new hal should fix it.

B--B, 2016-06-20:
@dookiedude yeah I agree with you! We released yesterday evening and no one reported this bug until now so hopefully it's finally fixed

closed 2016-08-09 by B--B