AOSP-JF-MM/issue_tracking_aosp #70: ROM Update changed lockscreen from 4x4 to 3x3

tophee, 2016-05-24:
I just updated by ROM to AOSP-6.0-20160416-jflte (on top of an earlier version the same ROM) according to the instructions provided [here](http://forum.xda-developers.com/galaxy-s4/i9505-orig-develop/jdcteam-android-source-project-mra58v-t3251663/post63887420#post63887420). Since this update, I am unable to get passed the lockscreen because it is now giving me a 3x3 pattern lockscreen while my original pattern was a 4x4 pattern.

I'm not sure if anyone will provide assistance here or whether I will have to post this as a question elsewhere. Please let me know.

FernBch, 2016-05-24:
Did you make a nandroid to revert to? If so restore it. Since there have been so many changes tou may have to clean flash to get the latest version. Dirty flashing doesn't work well for everyone, me included.

tophee, 2016-05-25:
I have partial nandroid backup which includes only the boot.img, recovery.img and system files. I ran out of space while doing the data backup. But since my data is still there and untouched, I guess boot, recovery, and system will do, right?

I have never restored a nandroid backup and I have also never downgraded to an older ROM version. So I'm not sure whether I can just restore these on top of the new ROM version...

B--B, 2017-03-12:
Issue caused when we moved from AOSP to CAF code, can be closed

closed 2017-03-12 by B--B