AOSP-JF-MM/issue_tracking_aosp #73: No audio after reccuring bluetooth bug.

codeseven58, 2016-06-14:
Phone model: Sgh-m919
Baseband version: ex. m919uvufok3
ROM version: ex. 20160612
GApps Package: Open GApps pico 20160612

Did you do a clean install for the currently installed build? yes 

Did you reboot after having the issue? no
Did you do any system modification? no
Did you flash another kernel? no
Did you install Xposed framework? no

Known bug: while listening to music via bluetooth and the screen is off, bluetooth bugs and stops playing audio after changing songs.

My bug: after having this bug and then disconnecting from bluetooth (leaving bluetooth on), my phone would not play any audio out of the external speaker. the audio began playing again after i forcably closed my music player (pandora). I did not test for in call audio on either speaker option.

http://pastebin.com/VV5bkfTV

codeseven58, 2016-06-15:
also. I do have adaway installed and gsm battery monitor installed. both i guess did modify the system. not sure if they can interfere or what.

B--B, 2016-06-18:
No it was something missing in the kernel, we are testing now and seems working without any issue, see the discussion here: https://github.com/AOSP-JF-MM/android_hardware_qcom_audio/commit/58db0376cacc555cb947ae649062a20b955ec8cf 
And this is the commit: https://github.com/AOSP-JF-MM/android_kernel_samsung_jf/commit/f9a4b8193a584cdfe807e86fa76855c2d0c4d86f

B--B, 2016-06-18:
Our testers reported that latest kernel changes and the new audio HAL fixes this issue, so this can be closed 

Related bugs:
https://github.com/AOSP-JF-MM/issue_tracking_aosp/issues/71
https://github.com/AOSP-JF-MM/issue_tracking_aosp/issues/30
https://github.com/AOSP-JF-MM/issue_tracking_aosp/issues/61

closed 2016-06-18 by B--B

codeseven58, 2016-06-19:
Woohoo! :-D