AOSP-JF-MM/issue_tracking_aosp #74: Android.Process.Media has stopped.

codeseven58, 2016-06-16:
Phone model: Sgh-m919
Baseband version: ex. m919uvufok3
ROM version: ex. 20160612
GApps Package: Open GApps pico 20160612

Did you do a clean install for the currently installed build? yes

Did you reboot after having the issue? no
Did you do any system modification? no
Did you flash another kernel? no
Did you install Xposed framework? no

plugged my phone into my computer, selected file transfer. got media has stopped FC

B--B, 2016-06-18:
Weird, i never had a crash on my Kubuntu machines, only thing that i added on ramdisk is the adb entry (otherwise we can't debug during the boot phase), i tried to add this https://github.com/AOSP-JF-MM/platform_device_samsung_jflte/commit/693e150388dbf552cc7290d461ac3774c5ed8288 so the usb is disabled between switch to another state, now i'm waiting for some report from our testers! Everything else should be fine, i've checked the code and we are synced with cm, they do not have this issue so i suppose that the problem was there

codeseven58, 2016-06-19:
Awesome. Im syncing the manifest right now and will be testing the code by the end of the day myself.

codeseven58, 2016-06-21:
seems im still having this issue. here is another log file 

http://pastebin.com/ktyYPdab

timerons, 2016-07-04:
i had the same issue. now the bug is gone. i am not sure what did the trick.

i moved data on the card and back and on the card again.
i granted some process access to the card

when you check my bug report, you can see that was a access right issue

https://github.com/AOSP-JF-MM/issue_tracking_aosp/issues/66

codeseven58, 2016-07-07:
wanted to add, not sure if it means anything, but mass storage works perfectly fine when in TWRP

codeseven58, 2016-07-11:
july 9th release is still giving me an error. 

http://pastebin.com/QSE6Ybnk

codeseven58, 2016-07-13:
Possible clue. I installed the drivers for my SGH-M919 from the samsung website and tried again. Process stopped again. but a quick moment later it connected without the process stopping. so i selected charging and then MTP again and got the process stopped message. but this time i switched to PTP and my phone connected immediately. then i switched back to MTP and it connected immediately. 

Is my phone trying to communicate with windows while it's still installing the drivers or initializing them? maybe that's what's happening?

B--B, 2016-07-14:
That' s weird, i' m not able to crash Media process, i've tried several times on kubuntu, opensuse and windows 7 (windows tested in virtual machine).. Please try to disable usb debugging and try to connect the device, let's see if helps

codeseven58, 2016-07-14:
I haven't tried yet on anything else but windows 10. that's probably the issue right there. i did try again with usb debugging  disabled, but it still gave me the error. I think it is because windows 10 is taking  a little longer to initiate the phone. cause after it's done initiating, i can use MTP or PTP to my hearts desire. 

is there a way to delay any communications between the phone and windows 10 until the initialization is completed?

codeseven58, 2016-08-18:
Everything seems to be working fine with the 13/08 release.

closed 2016-08-18 by codeseven58