AOSP-JF-MM/issue_tracking_aosp #76: Lag when scrollung up/down

FernBch, 2016-06-19:
Device: SCH-I545 OF1 modem
Clean Flash of AOSP-6.0-20160615-jflte.zip
Np system mods.

When scrolling up/down in any open app theere are lags, stutters, etc on the display. Spmetimes lasts as long as 2-3 seconds.

Logcat:

https://www.dropbox.com/s/ixtd6sf50u2q3x3/Log_2016-06-18_20-36-12.txt?dl=0

B--B, 2016-06-19:
Try the latest release (18), it's working fine for me

FernBch, 2016-06-20:
Still present but isn't as pronounced.

https://www.dropbox.com/s/3550omllioaffar/Log_2016-06-19_19-25-39.txt?dl=0

I may make some changes in STweaks and see if it makes an improvement.

FernBch, 2016-06-24:
I flaahed 21/06 today. Only been up and running for about an hour. Looks good so far. I'll report back in the next 24 hours.

FernBch, 2016-06-24:
The issue has improved but s still present.

https://www.dropbox.com/s/k19hip3kjfyhpk4/Log_2016-06-24_09-33-55.txt?dl=0

FernBch, 2016-07-05:
Flashed July 3 build. Issue persists.

New Logcat

https://www.dropbox.com/s/nebg9jrt48d7vd3/Log_2016-07-05_17-15-08.txt?dl=0

FernBch, 2016-07-11:
Still present in 09/07 release

B--B, 2016-07-14:
No one reported this issue in 09/07 public release, so probably it's something related to your device

WindowManager: Screen frozen for +550ms due to Window{828bc3e u0 com.teslacoilsw.launcher/com.teslacoilsw.launcher.NovaLauncher}

Your log is full of this, it's NovaLauncher your problem, but i know that @smeroni68 use Nova too and does not have this issue..

FernBch, 2016-07-14:
Ok. I guess I'll go to Tespacoil for bug reporting on this.

Thanks

closed 2016-07-14 by FernBch