AOSP-JF-MM/issue_tracking_aosp #80: Language & Input - Minor bug found

FernBch, 2016-06-24:
In the keyboard language settings English - Singapore, English - UK, and Englis - US are listed twice.

[IMG]http://i1075.photobucket.com/albums/w434/FernBchJeep/Mobile%20Uploads/Screenshot_20160623-185249_zps019slidf.png[/IMG]

FernBch, 2016-07-05:
I actually see doubles of every choice for English Language selection.

B--B, 2017-04-03:
This issue belongs to MM, but next release will be the last, so i close this as MM reached the EOL, N is our stable release now, and unfortunately i have no time to keep two different ROMs alive for multiple devices, BTW with N stable and bug free (wifi hotspot will be fixed in next upcoming release) i don't see the sense to stay with an older android release and OMS working only in legacy mode

closed 2017-04-03 by B--B