AOSP-JF-MM/issue_tracking_aosp #83: No longer able to update SU binary by "Normal" method.

FernBch, 2016-07-11:
Flaahed 09/07 release today and I am no longer able to update the SU binary by the normal method. The only option that works now is via TWRP. This was not present in the 03/07 test build.

https://www.dropbox.com/s/4262kqohy4qrcdz/Screenshot_20160710-174128.png?dl=0

Logcat

https://www.dropbox.com/s/q92eqjuiedvr6ul/Log_2016-07-10_17-42-00.txt?dl=0

B--B, 2016-07-14:
No one reported this issue in public release, you probably messed something during the update, and you should not need to update the binary, the rom already have the latest supersu version (2.76)
See here: https://github.com/AOSP-JF-MM/platform_vendor_aosp/commit/dc176c90fe1d4270634c1686be32562f588c1309

FernBch, 2016-07-14:
Interesting. I have always been asked by SuperSU to update the binary after installing regardless of which ROM I was currently running. On the 03/07 build I could update normally but on the latest it can only be done via TWRP. I wonder if this is specific to the SCH-I545. The versions of SU included in the ROM after 2.52 fail during installation. When I flash a new build I also have to flash 2.52 via recovery and allow SU to update from Play or it does not work.

B--B, 2017-03-12:
Fixed, this report can be closed, now root is systemless for everyone and for all variants

closed 2017-03-12 by B--B