AOSP-JF-MM/issue_tracking_aosp #84: Voice input doesn't work with messenger app.

ke4clu, 2016-07-20:
GApps Package: Open GApps PICO

Did you do a clean install for the currently installed build? yes ( wiped with TWRP )
Did you reboot after having the issue? yes
Did you do any system modification? yes: no
Did you flash another kernel? yes: stock kernel available on RomHut
Did you install Xposed framework? no
Kernel Version:  3.4.112-Alucard-PWR-CORE=APSP-6.0 b-b Jul 9 20:34:49 CEST 2016

Thanks.  I installed the above release and all my Bluetooth issues are resolved.   There is however on issue with SMS.   When sending messages I can not use voice input.  I tap the microphone on the keyboard and nothing happens.  Not a deal beaker but a small annoyance.

B--B, 2016-07-20:
We enabled again fluence on latest beta builds so actually I can't test with my device the public release, but should work even in public build.. Have you granted the "use mic" permission to messenger app?

ke4clu, 2016-07-20:
Thanks. That took care of it.

closed 2016-08-09 by B--B