AOSP-JF-MM/issue_tracking_aosp #85: Battery drops really quickly...

zabihchaudhry, 2016-07-25:
The battery drainage increases after installing this rom. 50% gone in 1 hour. So worried... Everything is off like scanning and nfc....

codeseven58, 2016-08-18:
The first boot after install ALWAYS runs power hungry. You have to fully charge it at least once for the OS to sort of calibrate itself

zabihchaudhry, 2016-08-18:
Thanks fpr the reply. But its not only for the first time, it drops battery even on standby..

codeseven58, 2016-08-21:
you may have a dead cell in your battery then. get a new one. or you may have a short in your battery. it happens a lot in car batteries. and i do mean A LOT

zabihchaudhry, 2016-08-21:
[image]
 This is new battery i bought. I only charge my phone on 2amp charger. See what is happening... :(

smeroni68, 2016-08-21:
Did you see that the phone is awake full time? There is an app or process that prevent deepsleep. Install BetterBatteryStats and check the responsable of this behaviour.

zabihchaudhry, 2016-08-21:
These are the screenshots from batterystats... I don't know how to use this
app and how to control apps eating battery. Can you please have a look and
give me a solution to that? Secondly, every scanning is off and wifi stays
on even if it is off. I don't know why

B--B, 2016-08-21:
This is a bug report, not an how to, sorry but there's an xda thread for these kind of questions, if some process is keeping your phone awake and happens only to you then it's not a rom bug, probably you have some rogue app that blocks deep sleep

closed 2016-08-21 by B--B