AOSP-JF-MM/issue_tracking_aosp #86: Notification stops working after whatsapp call

zabihchaudhry, 2016-07-30:
Whenever i receive call from facebook or whatsapp then my notification stops working except vibration. Always had to restart. Please fix it.

FernBch, 2016-07-30:
Clean flash or dirty?

A log would be needed here. Without it one can only guess at a possible cause.

zabihchaudhry, 2016-07-30:
Clean install... How can i show you the logs

FernBch, 2016-07-30:
Install an app such as logcat extreme, catlog, syslog, etc. from Play, on first use enable SU access, and save the log. The steps to save vary depending in which app you install but it is quite easy. Once saved upload it to Dropbox, Pastebin, etc. and then post the link to the log here.

zabihchaudhry, 2016-07-30:
Thats my link for log:
https://www.dropbox.com/s/hyktqu7dzs75rvy/Log_2016-07-31_05-37-47.txt?dl=0

B--B, 2017-03-12:
Fixed, this report can be closed

closed 2017-03-12 by B--B