AOSP-JF-MM/issue_tracking_aosp #87: Pandora Tries to load but crashes.

ke4clu, 2016-08-03:
Phone model: M919
Baseband version: M919UVUFOK3
ROM version: AOSP-6.0-20160726-jflte.zip
GApps Package: Open GApps Nano 20160803

Did you do a clean install for the currently installed build? yes

Did you reboot after having the issue? yes
Did you do any system modification? no
Did you flash another kernel? no
Did you install Xposed framework? no

Pandora tries to load but crashes and exits.   Work flawlessly with AOSP-6.0-20160709-jflte.zip

codeseven58, 2016-08-04:
I can verify this. here's my log. http://pastebin.com/n8wVEK07 . seems like the issue is with a missing class or file? line 250

> Caused by: java.lang.ClassNotFoundException: Didn't find class "android.webkit.ServiceWorkerController" on path: DexPathList[[zip file "/system/app/webview/webview.apk"],nativeLibraryDirectories=[/system/app/webview/lib/arm, /system/app/webview/webview.apk!/lib/armeabi-v7a, /vendor/lib, /system/lib]]

ke4clu, 2016-08-04:
Thanks for the quit response. 
Same here with Pandora will start to play unless I close the error popup. 
So what's the remedy?

ke4clu, 2016-08-04:
OK.  After browsing my device I do find Webview at /system/app/webview/webview.apk is there.

ke4clu, 2016-08-05:
I have the same issue with Pandora starting to play until I close the error box.   I don't know how long it would play because I always close the box:)

codeseven58, 2016-08-15:
2016/08/13 seems to be running fine for me. I haven't had any issues at all whatsoever in fact. I would suggest closing this issue if Ke4clu isn't having issues as well.

ke4clu, 2016-08-15:
codeseven58 ,
You were having the same issue as me.  What did you do to resolve it.

ke4clu, 2016-08-15:
I had back rev-ed to 2016-07-09 until I read you post.   I restored a backup of 2016-07-29 and the problem persists.

smeroni68, 2016-08-15:
Did you have files on external sd? If yes, have you tried to backup your files and format the sd?

B--B, 2016-08-15:
Update the ROM to latest public release, not 2016/07/29 version

ke4clu, 2016-08-15:
Link to latest public please.

FernBch, 2016-08-15:
It's in the OP on xda, as always.

https://romhut.com/roms/aosp-jf-6-0

ke4clu, 2016-08-16:
Thanks for the information guys.   After applying the recommendation from BB--BB and installing 20160812 all is well.

closed 2016-08-16 by ke4clu